YanjieZe/3D-Diffusion-Policy #2: MuJoCo compilation issues

dennisushi, 2024-03-07:
Whenever I run the policy training script (eg `bash scripts/train_policy.sh diffusion_image metaworld_push_image 0112 0 0 ` but also any combination) I get mujoco compilation errors. Have you encountered this?

```bash
Compiling /home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx
performance hint: /home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_warning_callback' to allow an error code to be returned.
performance hint: /home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx:104:5: Exception check on 'c_error_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_error_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_error_callback' to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

/home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to the type of 'c_warning_callback'.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to the type of 'c_error_callback'.
Error executing job with overrides: ['task=metaworld_push_image', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=metaworld_push_image-diffusion_image-0112', 'logging.mode=online', 'checkpoint.save_ckpt=False']
Traceback (most recent call last):
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/utils.py", line 639, in _locate
    obj = getattr(obj, part)
AttributeError: module 'diffusion_policy_3d.env_runner' has no attribute 'metaworld_image_runner'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/utils.py", line 645, in _locate
    obj = import_module(mod)
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/metaworld_image_runner.py", line 6, in <module>
    from diffusion_policy_3d.env import MetaWorldEnv
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/__init__.py", line 2, in <module>
    from .adroit import AdroitEnv
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/adroit/__init__.py", line 1, in <module>
    from .adroit import AdroitEnv
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/adroit/adroit.py", line 12, in <module>
    import mj_envs
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/third_party/rrl-dependencies/mj_envs/mj_envs/__init__.py", line 1, in <module>
    import  mj_envs.hand_manipulation_suite 
[... 48 lines cut ...]
    raise ex
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 41, in main
    workspace.run()
  File "/home/dennisushi/repos/3dp/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/workspace/train_diffusion_unet_hybrid_workspace.py", line 111, in run
    env_runner = hydra.utils.instantiate(
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 325, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET), full_key)
  File "/home/dennisushi/micromamba_envs/envs/diffusion_policy_v2/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 139, in _resolve_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error locating target 'diffusion_policy_3d.env_runner.metaworld_image_runner.MetaworldImageRunner', see chained exception above.
full_key: task.env_runner

```

dennisushi, 2024-03-07:
Soloved by downgrading cython [link](https://github.com/openai/mujoco-py/issues/773#issuecomment-1639684035)

closed 2024-03-07 by dennisushi

YanjieZe, 2024-03-08:
Nice!