YanjieZe/3D-Diffusion-Policy #4: Image to 3D point cloud?

DrBwts, 2024-03-12:
How are you converting your 2D camera images to 3D point clouds?

You don't go into any detail about this in your paper or provide a reference for the method. I was wondering if you could either elaborate or point me to the code that deals with this method?

YanjieZe, 2024-03-12:
Depths could be directly converted into point clouds with camera parameters. It is not a method we propose or introduced by others, but one basic truth in 3D vision.

There are some online blog that introduce this, such as [this](https://medium.com/yodayoda/from-depth-map-to-point-cloud-7473721d3f).

closed 2024-03-12 by YanjieZe

DrBwts, 2024-03-18:
I checked out your link & I'm not convinced you can calculate depth like that as equation 1 is not unique.