YanjieZe/3D-Diffusion-Policy #5: bash scripts/eval_policy.sh dp3 adroit_hammer_pointcloud 0112 0 0报错AttributeError: 'ParameterDict' object has no attribute 'point_cloud'

HIT-ZY, 2024-03-22:
在跑Evaluate a saved policy or use it for inference. For example:bash scripts/eval_policy.sh dp3 adroit_hammer_pointcloud 0112 0 0这一句时出现报错Traceback (most recent call last):
  File "eval.py", line 34, in <module>
    main()
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "eval.py", line 31, in main
    workspace.eval()
  File "/data1/zhouyi/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/workspace/train_diffusion_unet_hybrid_pointcloud_workspace.py", line 361, in eval
    runner_log = env_runner.run(policy)
  File "/data1/zhouyi/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/adroit_pointcloud_runner.py", line 94, in run
    action_dict = policy.predict_action(obs_dict_input)
  File "/data1/zhouyi/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/policy/diffusion_unet_hybrid_pointcloud_policy.py", line 192, in predict_action
    nobs = self.normalizer.normalize(obs_dict)
  File "/data1/zhouyi/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/model/common/normalizer.py", line 69, in normalize
    return self._normalize_impl(x, forward=True)
  File "/data1/zhouyi/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/model/common/normalizer.py", line 59, in _normalize_impl
    params = self.params_dict[key]
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/torch/nn/modules/container.py", line 738, in __getitem__
    return getattr(self, attr)
  File "/data1/zhouyi/anaconda3/envs/dex/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1614, in __getattr__
    raise AttributeError("'{}' object has no attribute '{}'".format(
AttributeError: 'ParameterDict' object has no attribute 'point_cloud'
前面的过程均正常，到这一步报错，请问应该怎么fix

YanjieZe, 2024-03-22:
hi，你的training可以正常跑吗

HIT-ZY, 2024-03-22:
training应该是正常的，就是evaluate报这个错，前面的包括环境配置应该都正常

YanjieZe, 2024-03-22:
我这边eval能正常跑。你有确定你eval的command和training用的是一样的算法/环境以及一样的additional info吗
[image]

YanjieZe, 2024-03-22:
以及train_policy.sh里，save_ckpt有设置为True吗

HIT-ZY, 2024-03-22:
算法/环境应该是一样的吧，我跑的就是README里面给出的两个命令：
bash scripts/train_policy.sh dp3 adroit_hammer_pointcloud 0112 0 0
bash scripts/eval_policy.sh dp3 adroit_hammer_pointcloud 0112 0 0
前一个正常，后一个报以上错误

YanjieZe, 2024-03-22:
你有在train_policy.sh里把save_ckpt设置为True吗

HIT-ZY, 2024-03-22:
train_policy.sh里之前save_ckpt没有设置为True，我设置为True试一下

YanjieZe, 2024-03-22:
没有设置为True的话，你training的ckpt是不会保存的，所以你eval的时候就没有load ckpt :)

HIT-ZY, 2024-03-22:
所以才会报AttributeError: 'ParameterDict' object has no attribute 'point_cloud' 是吗

YanjieZe, 2024-03-22:
应该是

HIT-ZY, 2024-03-22:
好的谢谢大佬

YanjieZe, 2024-03-22:
客气

closed 2024-03-22 by YanjieZe