YanjieZe/3D-Diffusion-Policy #16: data size about train

1223haohao, 2024-04-07:
Thanks for your excellent work!
I can't get the same results for a few tasks.So I want to ask how much episodes that you have used in your experiment!
Is it ten or more?

YanjieZe, 2024-04-07:
Hi, thank you!

There are variations between generated demonstrations, since we only use 10 demonstrations for most tasks. When demonstrations are different, the results could be different. This is one main reason why we experiment on a lot of tasks.

We are also preparing to release a set of demonstrations that could produce some fixed results. In our paper, we keep all algorithms using the same demonstrations for fairness.

For your own experiments, if you find that the results are not good, maybe re-generate the demonstrations and train again.

Hope this helps:)

1223haohao, 2024-04-10:
> Hi, thank you!
>
> There are variations between generated demonstrations, since we only use 10 demonstrations for most tasks. When demonstrations are different, the results could be different. This is [...]

Thank you for your answer. I have successfully reproduced the results in the paper through multiple attempts to regenerate the data.

But I'm sorry to bother you again. I have another question to ask.

Why does the seed remain unchanged, but there are still differences in each generated demonstration, as the random seed has already been set to 0 by default in the gendemonstration_expert.py file?

Is there a way for me to generate the same data every time?

YanjieZe, 2024-04-10:
Good question. It is a well-known issue that the simulation would give same results even if using the same seed. I think it is caused by the simulation itself, such as mujoco, isaacgym, etc. These simulators are mostly written in C /C++ and wrapped into python. Thus only fixing randomness in python is not enough.

closed 2024-04-17 by YanjieZe