YanjieZe/3D-Diffusion-Policy #17: How do I fix it?

SshiJwu, 2024-04-08:
> (diffusion_rl) (base) user@user-System-Product-Name:~/Downloads/diffusion_rl$ bash scripts/gen_demonstration_dexart.sh laptop
Data already exists at ../../3D-Diffusion-Policy/data/dexart_laptop_expert.zarr
If you want to overwrite, delete the existing directory first.
Do you want to overwrite? (y/n)
Overwriting ../../3D-Diffusion-Policy/data/dexart_laptop_expert.zarr
using seen instances [11395, 11405, 11406, 11477, 11581, 11586, 9996, 10090, 10098, 10101, 10125]
device: cuda
Traceback (most recent call last):
  File "examples/gen_demonstration_expert.py", line 236, in <module>
    main()
  File "examples/gen_demonstration_expert.py", line 75, in main
    env = create_env(task_name=task_name,
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/create_env.py", line 37, in create_env
    env = LaptopRLEnv(**env_params, friction=5)
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/rl_env/laptop_env.py", line 20, in __init__
    super().__init__(use_gui, frame_skip, friction=friction, index=index, **renderer_kwargs)
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/sim_env/laptop_env.py", line 15, in __init__
    super().__init__(use_gui=use_gui, frame_skip=frame_skip, **renderer_kwargs)
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/rl_env/base.py", line 32, in __init__
    super().__init__(use_gui=use_gui, frame_skip=frame_skip, use_visual_obs=use_visual_obs, **renderer_kwargs)
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/sim_env/base.py", line 25, in __init__
    engine, renderer = get_engine_and_renderer(use_gui=use_gui, need_offscreen_render=need_offscreen_render,
  File "/home/user/Downloads/diffusion_rl/third_party/dexart-release/dexart/env/sim_env/constructor.py", line 36, in get_engine_and_renderer
    _renderer = sapien.SapienRenderer(default_mipmap_levels=mipmap_levels, offscreen_only=not use_gui,
RuntimeError: Cannot find any cuda device suitable for rendering.

Mon Apr  8 22:07:39 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.161.07             Driver Version: 535.161.07   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3090        Off | 00000000:01:00.0  On |                  N/A |
| 30%   48C    P8              32W / 350W |    699MiB / 24576MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1026      G   /usr/lib/xorg/Xorg                           53MiB |
|    0   N/A  N/A      1656      G   /usr/lib/xorg/Xorg                          235MiB |
|    0   N/A  N/A      1785      G   /usr/bin/gnome-shell                         50MiB |
|    0   N/A  N/A      2273      G   ...ures=SpareRendererForSitePerProcess       16MiB |
|    0   N/A  N/A     85202      G   ...sion,SpareRendererForSitePerProcess      150MiB |
|    0   N/A  N/A     85677      G   /usr/lib/firefox/firefox                    156MiB |
+-----------------------------------------------------------nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Tue_May__3_18:49:52_PDT_2022
Cuda compilation tools, release 11.7, V11.7.64
Build cuda_11.7.r11.7/compiler.31294372_0----------------------------+

YanjieZe, 2024-04-09:
Hi, it seems this issue is similar to yours: [here](https://github.com/haosulab/ManiSkill2/issues/92)

SshiJwu, 2024-04-09:
I run `vulkaninfo` and there are no errors.

SshiJwu, 2024-04-11:
Hello, can you release some demonstrations about dexart?

YanjieZe, 2024-04-11:
Hi, we will release some demonstrations for benchmarking recently. Please stay tuned. Thank you very much!

YanjieZe, 2024-04-11:
If you could not install the dexart env, maybe try the authors' original codebase first: [here](https://github.com/Kami-code/dexart-release). Or maybe change a machine you are using to see whether it is caused by your computer.

closed 2024-04-17 by YanjieZe

Kimagure7, 2024-06-03:
> hi, have you solved it? I got the same error!

see [this](https://github.com/haosulab/SAPIEN/issues/115) for reference

xiaolijz, 2024-10-15:
> Hello, can you release some demonstrations about dexart?

I encountered the same problem, have you solved it?

DavidLXu, 2024-12-15:
> > Hello, can you release some demonstrations about dexart?
> 
> I encountered the same problem, have you solved it?

try this command
```
export VK_ICD_FILENAMES=/etc/vulkan/icd.d/nvidia_icd.json
```

b-karthiknair, 2025-01-09:
Also, ensure that the `CUDA_VISIBLE_DEVICES` environment variable is set to a valid GPU ID (`2` is the default for this script).