YanjieZe/3D-Diffusion-Policy #21: About dexart benchmark

seagul1, 2024-04-26:
Hello!
when I train DP3 under dexart bucket, the result shows the success rate on train env is lower than test env, Is that normal? And i want to know which success rate you use in your paper?

YanjieZe, 2024-04-26:
Yes, it is normal. The `test env` there means using some test objects not seen in the training environment. This environment is originally provided by the dexart paper. In our paper, we only study the training environment, since in the test environment the results are just generally lower and do not show other new information.

We might consider removing this part in the code in future, if it leads to confusion.

seagul1, 2024-04-28:
Thanks for your prompt reply！

closed 2024-04-28 by seagul1