YanjieZe/3D-Diffusion-Policy #22: Reproduce results

zhouzheyuan, 2024-05-01:
Hi Yanjie, thanks for your great work! 

Currently, I'm reproducing the results of DexArt and I'm noticing that the results seem to be heavily influenced by the generated demonstration. 
[image]
Specifically, Exp. 0322 and Exp. 0430 used the same demonstration, where Exp. 0430 was repeated five times using different seeds. Exp. 0501 regenerated the demonstration.

It can be seen that the experimental results for the different demonstrations are varied and both are lower than the results reported in the paper. I wonder if there is a way to stabilized reproduction results.

YanjieZe, 2024-05-01:
Hi, how many demonstrations do you use? In DexArt we actually use 100 demonstrations (see Table III). 

I think the instability is led by the small number of demonstrations. We would consider releasing a few demos so to help better stabilize results.

YanjieZe, 2024-05-03:
Hi, thank you for reminding:) I have changed that

closed 2024-05-03 by zhouzheyuan