YanjieZe/3D-Diffusion-Policy #25: Episode collection FPS, processing robot delays

rakhimovv, 2024-05-07:
Hi, thank you for your wonderful work!

I have a question regarding the FPS during data collection, specifically the time interval between two neighboring frames.

I'm asking because the model utilizes historical observations, particularly point clouds. I assume that if the change between two neighboring frames is less than 1 second, there won't be significant differences in the observations. Do we use the point cloud history primarily to address occlusions, in cases where an object is not visible in the current frame but was visible in previous frames?

Additionally, real data collection is not always perfect, and sometimes the robot may move very slowly or even stop for 1-2 seconds due to human factors. Should we preprocess this data, or would it be fine to use the data as it is?

I would greatly appreciate your answer.

YanjieZe, 2024-05-07:
Hi, the data is collected as a MDP, so that the time interval between steps during data collection actually could be any.
For example, if you make the robots move one step forward, and then you go to grab a coffee, and then you go back to continue the data collection, this would not effect your final data, since they are collected step by step, and executed as such also.

rakhimovv, 2024-05-07:
Sounds good, thank you!

closed 2024-05-07 by rakhimovv