YanjieZe/3D-Diffusion-Policy #26: End-effector state class imablance

rakhimovv, 2024-05-07:
Hi, thank you for your wonderful work!

Have you experienced problems with class imbalance?

For example, if I want to collect episodes for a "pick" task (e.g., "pick a cube"), the "closed" state would only occur in about 1 out of every 50 frames. My initial assumption is that it might be easier for the model to always predict the "open" state.

I realize this issue is somewhat general, but I would greatly appreciate your advice. For instance, I noticed that there is no class weighting or special sampling mode in the code.

YanjieZe, 2024-05-07:
Yes, so that the conditioning is important for the right prediction. 
For example, the robot predicts the open state when it moves toward the object; predicts the close state when it grasps the object/

rakhimovv, 2024-05-07:
Got it, thank you!

closed 2024-05-07 by rakhimovv