YanjieZe/3D-Diffusion-Policy #27: the results for assembly in Metaworld are far away from the paper

csufangyu, 2024-05-08:
Thank you very much for your excellent work, I ran the code to test the task in Metaworld , but the experimental results are far away from the results of the paper, here are my experimental results:
```
test_mean_score: 0.0                                                                                            
test_mean_score: 0.2                                                                                            
test_mean_score: 0.3                                                                                            
test_mean_score: 0.35
test_mean_score: 0.15                                                                                           
test_mean_score: 0.0                                                                                            
test_mean_score: 0.5                                                                                            
test_mean_score: 0.15                                                                                           
test_mean_score: 0.1                                                                                            
test_mean_score: 0.15                                                                                           
test_mean_score: 0.1
test_mean_score: 0.25                                                                                           
test_mean_score: 0.15
test_mean_score: 0.2                                                                                            
test_mean_score: 0.2
```
Would you like to know if there is anything that needs to be adjusted?

YanjieZe, 2024-05-08:
Hi, are you using our old code? in the current version we do not provide metaworld envs. Due to the small number of demonstrations, the results in Metaworld could have some variance. If you get some bad results, maybe try to regenereate demonstrations.

YanjieZe, 2024-05-08:
Hi, I have added more explanations in the readme. Thank you for the usage:)

csufangyu, 2024-05-08:
Yes, I use the old code. Thanks very much!

closed 2024-05-08 by csufangyu