YanjieZe/3D-Diffusion-Policy #30: Question about run in simulate env

xbkaishui, 2024-05-15:
Hi, thanks for your great work, I have a question, How to simulate training in a simulation environment? 
I can't find more details in code or document,  is Imitation learning does not necessarily require a simulation environment?

closed 2024-05-15 by YanjieZe

xbkaishui, 2024-05-16:
Thank you for your response. I am a beginner in the field of robotics and would like to ask how to set up a simulation environment. Do you have any recommended articles?

YanjieZe, 2024-05-16:
You could start by running some standard sim envs.  For example, ManiSkill: https://maniskill2.github.io/

xbkaishui, 2024-05-16:
ok thanks

zackzhao, 2024-06-07:
> Hi, simulation environments are not used for training, but only for evaluation. Imitation learning only trains a policy, but the imitation learning loss does not actually reflect how well the policy is. Therefore, simulation environments are needed.
> 
> I think this is very common practise in robotics, and is not we propose for the first time.

Hi, I also have a same question. Can we use human demonstration in sim envs? If yes , and can you give some suggestion how to use sim envs to collect data?