YanjieZe/3D-Diffusion-Policy #32: sim env for realdex_drill dataset

xbkaishui, 2024-05-24:
Hi， I can't found sim env for realdex_drill dataset,  is this impossible to sim this dataset? 
by the way,  how to code online inference logic?  

Thanks

YanjieZe, 2024-05-24:
Hi, realdex_drill dataset is only used in the real world. Currently it does not support simulation. 
The online inference logic is to first create your env and run the model in the env. You check the env runner for it: [here](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/adroit_runner.py)

xbkaishui, 2024-05-24:
I have another question,  in your real robot setup, you use Franka Robot,  currently I have a aubo robot. how to apply the trained model with aubo robot ?  should i retarget the  robot position of the action output ?

YanjieZe, 2024-05-24:
I think maybe you could collect your own dataset first by teleoperation, which would be easier to make it work. I am not sure whether retargering our data to your data would work, but you could definitely try it.

YanjieZe, 2024-05-24:
For your another question,
Our sim env is only used for evaluation.
DP3 is an offline imitation learning algorithm. It does need env interaction during training. Only for evaluation we deploy it in the sim env.

xbkaishui, 2024-05-24:
ok, can we talk in wechat? ^_^  how to retarget the robot position,  just world frame alignment ?

YanjieZe, 2024-05-24:
For discussions/technical questions, let us keep them in github issues/emails. Thank you:)

xbkaishui, 2024-05-24:
got it， I have three questions summary as below： 

1. it i want to apply the trained model with another robot , how to retarget the robot position, just world frame alignment ?
2. do you have some tips for  teleoperation arms with hand ?  what is  teleoperation tool do you use in your experiments ? 

thanks

YanjieZe, 2024-05-24:
1. Yes. you need to retarget the robot position to another robot. Our work does not study the cross embodiment, however.
2. Use keyboard to control arm, and use [dex-retargeting](https://github.com/dexsuite/dex-retargeting) to control hand. This has been illustrated in our paper.

xbkaishui, 2024-05-24:
got it. thanks

closed 2024-05-24 by xbkaishui