YanjieZe/3D-Diffusion-Policy #34: prediction_type for noise_scheduler

yquantao, 2024-05-28:
Hi,

if we consider the loss described in equation 2 in the paper, shouldn't [prediction_type in the config file](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/7ef163a6b8f74cdc2d387d53a46bf5e1088a2d9e/3D-Diffusion-Policy/diffusion_policy_3d/config/dp3.yaml#L50C5-L50C20) be epsilon? In the original diffusion policy code, it is set as epsilon.

YanjieZe, 2024-05-28:
Hi,

Yes, I think the loss in equation 2 describes the epsilon-type prediction. We do not detail the sample prediction in the paper for the simplicity, since this is not our core focus (but we have some related ablation experiments in the paper, you could possibly check)

yquantao, 2024-05-28:
Thank you for the reply. Maybe I didn't describe my question clearly.

In your released dp3.yaml config file, the default value of prediction_type for noise_scheduler is 'sample' now, which will make [target in this line](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/7ef163a6b8f74cdc2d387d53a46bf5e1088a2d9e/3D-Diffusion-Policy/diffusion_policy_3d/policy/dp3.py#L342) equal to original robot trajectory, rather than noise. Then it will not explain equation 2 in the paper. Or I miss something here?

yquantao, 2024-05-30:
I got it. But it is interesting that sample performs better than epsilon. According to stable diffusion, the advantage of epsilon prediction is that it often leads to better stability and training efficiency.