YanjieZe/3D-Diffusion-Policy #40: Could not initialize EGL, Failed to initialize OpenGL

Fisher-Wang, 2024-06-18:
After setting up my environment as instructed in INSTALL.md, when I run `bash scripts/gen_demonstration_adroit.sh hammer`, I got the following error:
```
Data already exists at ../../../3D-Diffusion-Policy/data/adroit_hammer_expert.zarr
If you want to overwrite, delete the existing directory first.
Do you want to overwrite? (y/n)
Overwriting ../../../3D-Diffusion-Policy/data/adroit_hammer_expert.zarr
Loaded expert ckpt from ../ckpts/vrl3_hammer.pt
/home/fs/cod/3D-Diffusion-Policy/third_party/gym-0.21.0/gym/logger.py:34: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize("%s: %s" % ("WARN", msg % args), "yellow"))
[MujocoPointcloudWrapper] use_point_crop: True
[MujocoPointcloudWrapper] sampling 512 points from point cloud using fps
Found 3 GPUs for rendering. Using device 0.
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
Could not initialize EGL
Traceback (most recent call last):
  File "gen_demonstration_expert.py", line 199, in <module>
    main()
  File "gen_demonstration_expert.py", line 91, in main
    time_step = env.reset()
  File "/home/fs/cod/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/mjpc_wrapper.py", line 205, in reset
    timestep = self.env.reset()
  File "/home/fs/cod/3D-Diffusion-Policy/third_party/VRL3/src/adroit.py", line 295, in reset
    obs_pixels, obs_sensor = self._env.reset()
  File "/home/fs/cod/3D-Diffusion-Policy/third_party/VRL3/src/rrl_local/rrl_multicam.py", line 162, in reset
    pixels, sensor_info = self.get_obs()
  File "/home/fs/cod/3D-Diffusion-Policy/third_party/VRL3/src/rrl_local/rrl_multicam.py", line 115, in get_obs
    img = self._env.env.sim.render(width=self.width, height=self.height, mode='offscreen', camera_name=cam, device_id=0) # TODO device id will think later
  File "mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
  File "mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
  File "mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```

My environment:
- Ubuntu 23.10
- CUDA 12.4
- NVIDIA Driver Version: 550.54.15

I understand this error may not be the problem of this repo, but I searched over the internet and found no solution. If anyone could help me, I would really appreciate it!

YanjieZe, 2024-06-21:
Hi, it seems there are some problems in your EGL/OpenGL library. But I also do not encounter this problem previously.

closed 2024-06-26 by YanjieZe

RealXuChe, 2024-10-07:
I encountered a similar issue, and the error message and stack trace are identical to yours.

In the end, I resolved the problem by configuring mujoco_py to use software rasterization. If you need mujoco_py to use software rasterization, you can modify this line https://github.com/YanjieZe/3D-Diffusion-Policy/blob/9fdfd0adcc7a9506118ad69b96ff7f4a425c8a40/third_party/mujoco-py-2.1.2.14/mujoco_py/builder.py#L79 by changing `LinuxGPUExtensionBuilder` to `LinuxCPUExtensionBuilder`. Additionally, before running, set the environment variable `MUJOCO_PY_FORCE_CPU=true`. In your case, after making the above modifications, you should use the command `MUJOCO_PY_FORCE_CPU=true bash scripts/gen_demonstration_adroit.sh hammer`.

Although I eventually resolved the issue using software rendering, some posts online and one of my colleagues indicated that they fixed the problem by updating the NVIDIA driver. My colleague didn't provide specific details, but you can refer to this: https://zhuanlan.zhihu.com/p/709337526.

However, I doubt that our issue was not caused by an outdated NVIDIA driver, since I see that your driver version is 550, which is relatively new. I suspect that we may not have fully installed all driver dependencies. We're both using Ubuntu, and as far as I know, Ubuntu has a rather aggressive package-splitting policy. Ubuntu developers tend to split a single program into multiple packages, so it's possible that we missed some of the packages related to the NVIDIA driver, leading to an incomplete installation. However, if you follow the method mentioned in [this article](https://zhuanlan.zhihu.com/p/709337526) and install the driver via the .run file, you'll likely get a full installation.

This is how I fixed the issue, along with other possible solutions. I noticed that quite some time has passed since you opened this issue, and I wonder if you've managed to resolve it. If so, I'd greatly appreciate it if you could share your solution.

euminds, 2025-08-24:
> LinuxGPUExtensionBuild

Do I need to reinstall mujoco after modifying the code?

RealXuChe, 2025-08-24:
> > I encountered a similar issue, and the error message and stack trace are identical to yours.
> > In the end, I resolved the problem by configuring mujoco_py to use software rasterization. If you need mujoco_py to use software rasterization, you can modify this line
> > [3D-Diffusion-Policy/third_party/mujoco-py-2.1.2.14/mujoco_py/builder.py](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/9fdfd0adcc7a9506118ad69b96ff7f4a425c8a40/third_party/mujoco-py-2.1.2.14/mujoco_py/builder.py#L79)
> > Line 79 in [9fdfd0a](/YanjieZe/3D-Diffusion-Policy/commit/9fdfd0adcc7a9506118ad69b96ff7f4a425c8a40)
> > Builder = LinuxGPUExtensionBuilder
> > by changing `LinuxGPUExtensionBuilder` to `LinuxCPUExtensionBuilder`. Additionally, before running, set the environment variable `MUJOCO_PY_FORCE_CPU=true`. In your case, after making the above modifications, you should use the command `MUJOCO_PY_FORCE_CPU=true bash scripts/gen_demonstration_adroit.sh hammer`.
> > Although I eventually resolved the issue using software rendering, some posts online and one of my colleagues indicated that they fixed the problem by updating the NVIDIA driver. My colleague didn't provide specific details, but you can refer to this: [zhuanlan.zhihu.com/p/709337526](https://zhuanlan.zhihu.com/p/709337526).
> > However, I doubt that our issue was not caused by an outdated NVIDIA driver, since I see that your driver version is 550, which is relatively new. I suspect that we may not have fully installed all driver dependencies. We're both using Ubuntu, and as far as I know, Ubuntu has a rather aggressive package-splitting policy. Ubuntu developers tend to split a single program into multiple packages, so it's possible that we missed some of the packages related to the NVIDIA driver, leading to an incomplete installation. However, if you follow the method mentioned in [this article](https://zhuanlan.zhihu.com/p/709337526) and install the driver via the .run file, you'll likely get a full installation.
> > This is how I fixed the issue, along with other possible solutions. I noticed that quite some time has passed since you opened this issue, and I wonder if you've managed to resolve it. If so, I'd greatly appreciate it if you could share your solution.
> 
> Do I need to reinstall mujoco after modifying the code?

https://github.com/YanjieZe/3D-Diffusion-Policy/blob/e15cdd00b6f088be1bcb9ef00f61e30a9df0239b/INSTALL.md?plain=1#L56-L61

No. If you followed the INSTALL.md, then you should have installed this dependency using `pip -e`. The `-e` option allows your modifications to take effect immediately without the need to reinstall.