YanjieZe/3D-Diffusion-Policy #41: How to visualize the gen_demonstration_expert?

kangkang2626, 2024-06-20:


YanjieZe, 2024-06-21:
Hi, thank you for your interest.
You could first load the demonstration (see code of dataset / data loader for reference, [here](https://github.com/YanjieZe/3D-Diffusion-Policy/tree/master/3D-Diffusion-Policy/diffusion_policy_3d/dataset)), and visualize the image sequence as video.

closed 2024-06-21 by YanjieZe

kangkang2626, 2024-06-21:
你好，我的意思是我现在通过脚本生成了.zarr的数据，里面是image、point cloud等字典信息，我想对其进行可视化应该怎么操作？还有就是dexart的演示生成可以通过云服务器吗?因为我现在有vulkan相关的报错。感谢您的回答。[svulkan2] [error] Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available

abcdsaltfish, 2024-09-24:
I see. The code below can show 1 picture. But how can I get a video?
```
from diffusion_policy_3d.dataset.adroit_dataset import AdroitDataset
import visualizer

dataset = AdroitDataset("3D-Diffusion-Policy/data/adroit_hammer_expert.zarr")

data = dataset[0]['obs']['point_cloud'].squeeze()
visualizer.visualize_pointcloud(data)
```

abcdsaltfish, 2024-09-24:
I encountered this error too.
```
Some required Vulkan extension is not present. You may not use the renderer to render, however, CPU resources will be still available
```
Command ```vulkaninfo``` shows 
```
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.70

/build/vulkan-UL09PJ/vulkan-1.1.70+dfsg1/demos/vulkaninfo.c:2700: failed with VK_ERROR_INITIALIZATION_FAILED
```
I don't know how to continue.