YanjieZe/3D-Diffusion-Policy #45: Data link of the real robot example data

jarvishou829, 2024-06-25:
Thanks for your great work. I notice that the ReadMe mentioned there is an example of real world data. But I cannot find it in the OneDrive and GoogleDrive link. Where can I get the access.

jarvishou829, 2024-06-25:
Ah, I just check the link on the github. I found it on the project page. BTW, how can I get the data before being encoded to the zarr format. I wanna record my own data, but have no idea on  the exact data structure.

YanjieZe, 2024-06-26:
Hi, thank you for your interest!
You could check the python script to generate simulation demonstrations first. It would provide you some basic understanding about the file structure and how to encode your own data.

closed 2024-06-26 by YanjieZe