YanjieZe/3D-Diffusion-Policy #46: point cloud

kingchou007, 2024-06-27:
Hi,

Thanks for the great work.

I am currently working on processing real robot data and have a question about point preprocessing. In the paper, it is mentioned that depth images are obtained with a size of 84 × 84 from a single camera, and then downsampling use the point 1024. I would like to know the original size of the images before running the preprocess_image function in the script.

I want to ensure that I am following the correct workflow. I have RGB images with dimensions (H, W, 3) and corresponding depth images with dimensions (H, W). My current approach involves resizing the depth images (run the process_image function in the scripts) and then creating the point cloud from these resized depth images.

However, after resizing, the depth images appear incorrect. If I use the raw depth (1280, 720), it looks fine. Could you clarify the proper method for processing the depth images and generating point clouds?

Thanks

YanjieZe, 2024-06-28:
Hi, thank you for your interest.

I think you could directly use your raw depth (so that the point cloud is right) and do the downsampling. Remember to do some crop before downsampling, to make the downsampled points more useful

closed 2024-06-28 by YanjieZe

Kimagure7, 2024-07-09:
> Hi, 你好，
> 
> Thanks for the great work.感谢您做的伟大工作。
> 
> I am currently working on processing real robot data and have a question about point preprocessing. In the paper, it is mentioned that depth images are obtained with a size of 84 × 84 from a single camera, and then downsampling use the point 1024. I would like to know the original size of the images before running the preprocess_image function in the script.我目前正在处理真实的机器人数据，并对点预处理有疑问。论文中提到，从单个相机获得尺寸为84×84的深度图像，然后使用点1024进行下采样。我想在运行中的preprocess_image函数之前知道图像的原始尺寸脚本。
> 
> I want to ensure that I am following the correct workflow. I have RGB images with dimensions (H, W, 3) and corresponding depth images with dimensions (H, W). My current approach involves resizing the depth images (run the process_image function in the scripts) and then creating the point cloud from these resized depth images.我想确保我遵循正确的工作流程。我有尺寸为 (H, W, 3) 的 RGB 图像和尺寸为 (H, W) 的相应深度图像。我当前的方法包括调整深度图像的大小（运行脚本中的 process_image 函数），然后从这些调整大小的深度图像创建点云。
> 
> However, after resizing, the depth images appear incorrect. If I use the raw depth (1280, 720), it looks fine. Could you clarify the proper method for processing the depth images and generating point clouds?但是，调整大小后，深度图像显示不正确。如果我使用原始深度（1280, 720），它看起来不错。您能否阐明处理深度图像和生成点云的正确方法？
> 
> Thanks 谢谢

How do you generate PointCloud data in real world data? Did you write the script yourself?

kingchou007, 2024-07-09:
> How do you generate PointCloud data in real world data? Did you write the script yourself?

Hi, 

You can take a look this [example](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/00d016103535b2d36956c65c0c59fa193dc448df/scripts/convert_real_robot_data.py#L38).

Kimagure7, 2024-07-09:
> Hi,
>
> You can take a look this [example](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/00d016103535b2d36956c65c0c59fa193dc448df/scripts/convert_real_robot_data.py#L38).

I haven't gotten to that point yet, I'm still struggling with generating point clouds from RGB-D pics.  Can you share the code about how you handle this? 2477036173@qq.com. 
BTW, have you succeed in deploying dp3 in a real robot?

laotanlaoban, 2024-12-17:
> 你好，
> 
> 感谢您的出色工作。
> 
> 我目前正在处理真实机器人数据，对点预处理有一个疑问。论文中提到，深度图像是从单个摄像头获取的，大小为 84 × 84，然后使用点 1024 进行下采样。我想知道在运行脚本中的 preprocess_image 函数之前图像的原始大小。
> 
> 我想确保我遵循正确的工作流程。我有尺寸为 (H, W, 3) 的 RGB 图像和尺寸为 (H, W) 的相应深度图像。我目前的方法是调整深度图像的大小（运行脚本中的 process_image 函数），然后从这些调整大小的深度图像创建点云。
> 
> 但是，调整大小后，深度图像看起来不正确。如果我使用原始深度（1280, 720），它看起来没问题。您能解释一下处理深度图像和生成点云的正确方法吗？
> 
> 谢谢

您好，请问您已经完成了真机部署吗，我在部署后，推理的策略和演示区别很大

kingchou007, 2024-12-18:
> 您好，请问您已经完成了真机部署吗，我在部署后，推理的策略和演示区别很大

之前在真机部署效果不太好，仿真不错。