YanjieZe/3D-Diffusion-Policy #48: question about real_robot_demo

kangkang2626, 2024-07-03:
hi, great work.
but i have a question, i use '  bash scripts/train_policy.sh dp3 realdex_drill 0112 0 0' to train the policy with the real robot data which you given, but with a zero success rate on all four real-world tasks, I visualized your expert demo as if it were incomplete, and I didn't modify your training code. So it's normal to have a zero success rate on a real task?
[image]

YanjieZe, 2024-07-03:
Hi, thank you for your interest.
The mean score is not success rate. It is the training error.
In fact, you can not get the success rate of a real-world task by the training code only, since you need to deploy your model in the real world and evaluate it.

closed 2024-07-03 by YanjieZe

kangkang2626, 2024-07-03:
ok, thanks

dennisushi, 2024-07-03:
@YanjieZe Further to this thread, is 0 the maximum test score?

I am getting almost 0 test score on real data (40 demos of a reaching task with panda, with actions represented as XYZ+RPY). When I evaluate the model on the real robot, it replicates the demonstrations quite well but misses the reaching target completely (It's a ball object).

kangkang2626, 2024-07-04:
@YanjieZe sorry to bother you. In addition, I visualized an expert demonstration of the drill task and found that the robotic arm did not pick up the drill bit and did not track the little green squares. Is this normal?

laotanlaoban, 2024-11-19:
> @YanjieZe进一步回答这个话题，0 是最高测试分数吗？
> 
> 我在真实数据（熊猫伸手任务的 40 个演示，动作表示为 XYZ+RPY）上几乎得不到测试分数。当我在真实机器人上评估模型时，它很好地复制了演示，但完全错过了伸手目标（它是一个球形物体）。

Hello, how can I deploy the training results to a real machine for observation?