YanjieZe/3D-Diffusion-Policy #49: DP3 baseline not generalizing.

dennisushi, 2024-07-04:
[image]
Hi, 
I am training the model in the simplest possible real task: reaching an object. This is done with 7DoF Panda controlled via end effector pose (world-frame XYZ-RPY 6 actions), training SimpleDP3 on 40 demonstrations (with default 0.02 validation ratio as in your configs) until the test score converges near 0. 

As you can see in the plot, one of the trajectories (which I believe is assigned as a validation trajectory) is incorrect. My main suspicion is that my data is non-Markovian as I do it continuously with real-time teleoperation and that the model cannot handle that (e.g. short pauses in the movement). 
If my suspicion is correct, I suspect a larger time horizon may fix it partially. 

Is the test score based on the validation trajectories as it doesn't match what I observe?
Is this overfitting to demonstrations normal for the model?

YanjieZe, 2024-07-05:
Hi, thank you for your interest!

Your experiments are actually interesting. As I observe in the figure, the line that you mention as not right, seems to be correct in its earlier part, but goes into messy in the end. Right? I am not pretty sure whether it is showing "not generalizing". Instead, I think it is more like that that trajectory is hard to predict (a sudden twist).

closed 2024-07-05 by YanjieZe

dennisushi, 2024-07-05:
The bad predictions start much before the twist (see rightmost part of rightmost plot), and that twist isn't sudden in time, just in space. It seems to be more of a generalization issue, as I do not observe task-oriented behaviour when new data is presented - which means either the data is insufficient or in some incorrect format.

kingchou007, 2024-07-16:
Hi @dennisushi

Did you evaluate dp3 in real world? I've tried implementing dp3 in real-world task, but I'm struggling to achieve any meaningful results. 

I don't know what am I missing or doing wrong that is leading to this extremely low success rate?

dennisushi, 2024-07-16:
@kingchou007 Hi, I tried to evaluate in real world. The behaviour I am getting is consistently bad. If you want we can chat and figure out where we are failing together.

kingchou007, 2024-07-17:
> @kingchou007 Hi, I tried to evaluate in real world. The behaviour I am getting is consistently bad. If you want we can chat and figure out where we are failing together.

sure, my email is jl3485@cornell.edu