YanjieZe/3D-Diffusion-Policy #51: question about data

kangkang2626, 2024-07-11:
Hi, I have some questions about the amount of data for each task. For example, in the real task 'drill', the shape of the point cloud is (2856, 1024, 3), which means that there are 2856 point clouds, but each expert demonstration of drill contains 29 frames, making a total of 40 demonstrations (but 29*40=1160). Similarly, the shape of img in the metaworld task of (2000, 128, 128, 3) seems to have nothing to do with the number of frames per demonstration . How is this data size determined?

moxqa, 2024-07-12:
> Hi, I have some questions about the amount of data for each task. For example, in the real task 'drill', the shape of the point cloud is (2856, 1024, 3), which means that there are 2856 point [...]

It seems like every demonstration have different length, you can read the zarr file to figure out the frame number of each episode. Well, it makes me curious with the 29 frames, how did that number come about:)

kangkang2626, 2024-07-12:
> It seems like every demonstration have different length, you can read the zarr file to figure out the frame number of each episode. Well, it makes me curious with the 29 frames, how did that number [...]

[image]
{
    "chunks": [
        100,
        84,
        84,
        3
    ],
    "compressor": {
        "blocksize": 0,
        "clevel": 3,
        "cname": "zstd",
        "id": "blosc",
        "shuffle": 1
    },
    "dtype": "|u1",
    "fill_value": 0,
    "filters": null,
    "order": "C",
    "shape": [
        2856,
        84,
        84,
        3
    ],
    "zarr_format": 2
}
like this , in fact, i am interested in the shape of 2856, how does that number come.

moxqa, 2024-07-12:
``
file_path = './3D-Diffusion-Policy/data/dexart_laptop_expert.zarr'

zarr_file = zarr.open(file_path, mode='r')
print(zarr_file.tree())
arr = zarr_file['meta/episode_ends']
print(arr[:])
``
run the code like this

moxqa, 2024-07-12:
```
file_path = './3D-Diffusion-Policy/data/dexart_laptop_expert.zarr'

zarr_file = zarr.open(file_path, mode='r')
print(zarr_file.tree())
arr = zarr_file['meta/episode_ends']
print(arr[:])
```
run the code like this

YanjieZe, 2024-07-28:
Hi, thank you for your interest.

Every episode has different number of frames, i.e.,  the length for each episode is different.

closed 2024-07-28 by YanjieZe