YanjieZe/3D-Diffusion-Policy #56: Request for update convert_real_robot_data.py

Charlie0257, 2024-07-27:
Hi, @YanjieZe 
Thanks for your great work!

I noticed that you updated the data in [real robot,](https://drive.google.com/file/d/1G5MP6Nzykku9sDDdzy7tlRqMBnKb253O/view?pli=1) which causes running **convert_real_robot_data.py** directly to report an error. Can you update **convert_real_robot_data.py** with the updated [real robot data](https://drive.google.com/file/d/1G5MP6Nzykku9sDDdzy7tlRqMBnKb253O/view?pli=1)?

YanjieZe, 2024-07-28:
Hi, thank you for your interest in our work.

Could you please describe the error in detail? The code and data we provide have been tested successfully and are not changed for a long time.

Charlie0257, 2024-07-28:
Thanks for your reply! @YanjieZe 

1. I noticed that you loaded datasets [realdex_roll](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/scripts/convert_real_robot_data.py#L87) and [realdex_roll.zarr](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/scripts/convert_real_robot_data.py#L88) in [convert_real_robot_data.py](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/scripts/convert_real_robot_data.py), but the real data link puts in drill_40demo_1024.zarr,  dumpling_new_40demo_1024.zarr, pour_40demo_1024.zarr and roll_40demo_1024.zarr:(
2. When I try to load dataset **drill_40demo_1024.zarr** in [convert_real_robot_data.py](https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/scripts/convert_real_robot_data.py), it reports an error because——
`FileNotFoundError: [Errno 2] No such file or directory: '../3D-Diffusion-Policy/data/drill_40demo_1024.zarr/meta/data.pkl'`

Anyway, thanks for your reply again:)
I'm sorry to take your time.

Charlie0257, 2024-08-01:
Hi, @blakery-star 
Could you update the corresponding data and codes?

blakery-star, 2024-08-02:
Hi Charlie,

Thank you for your question. Our script **convert_real_robot_data.py** serves as an example to demonstrate how you can preprocess your collected data and convert it into a format suitable for training. The real-world data we provide has already been preprocessed and is ready to use with the training script. This is **not the raw data**, and therefore, you should not use the **convert_real_robot_data.py** script to process it.

Charlie0257, 2024-08-02:
Thanks for your reply!

The problem now is that *convert_real_robot_data.py *mentioned in
*readme.md* is not working properly, can you maintain it?

Best,
Charlie

blakery-star, 2024-08-03:
@Charlie0257  What is your problem now? Our provided script convert_real_robot_data.py is used to demonstrate how to process the raw collected data. The main operations in the script are cropping using a bounding box and FPS downsampling. Our provided drill_40demo_1024.zarr has been processed and ready for training. So you **can't load it** using convert_real_robot_data.py. I recommend first you should collect real world demo and save the data as the format described in README. Then you can slightly modify our script convert_real_robot_data.py to load the raw data to process it. Then if you have further questions or need help, please let me know.

closed 2024-08-03 by blakery-star

laotanlaoban, 2024-11-19:
> @Charlie0257 What is your problem now? Our provided script convert_real_robot_data.py is used to demonstrate how to process the raw collected data. The main operations in the script are cropping [...]

Hello, after training with the real data you provided, how can I deploy it on my Franka robot arm to observe the effect of this strategy?