YanjieZe/3D-Diffusion-Policy #57: hydra.errors.InstantiationException: Error locating target 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner', see chained exception above. full_key: task.env_runner

Shi-jiayu, 2024-07-29:
When I run `bash scripts/train_policy.sh dp3 adroit_mammer 0112 0 0`, I encounter an error
Error executing job with overrides: ['task=adroit_hammer', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=adroit_hammer-dp3-0112', 'logging.mode=online', 'checkpoint.save_ckpt=True']
Traceback (most recent call last):
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 639, in _locate
    obj = getattr(obj, part)
AttributeError: module 'diffusion_policy_3d.env_runner' has no attribute 'adroit_runner'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 645, in _locate
    obj = import_module(mod)
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/shi/pku/2024.07/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/adroit_runner.py", line 5, in <module>
    from diffusion_policy_3d.env import AdroitEnv
  File "/home/shi/pku/2024.07/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/__init__.py", line 4, in <module>
    from .metaworld import MetaWorldEnv
  File "/home/shi/pku/2024.07/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/metaworld/__init__.py", line 1, in <module>
    from .metaworld_wrapper import MetaWorldEnv
  File "/home/shi/pku/2024.07/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/metaworld/metaworld_wrapper.py", line 10, in <module>
    from natsort import natsorted
ModuleNotFoundError: No module named 'natsort'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 134, in _resolve_target
    target = _locate(target)
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 648, in _locate
    raise ImportError(
ImportError: Error loading 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner':
ModuleNotFoundError("No module named 'natsort'")
Are you sure that 'adroit_runner' is importable from module 'diffusion_policy_3d.env_runner'?

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 503, in main
    workspace.run()
  File "train.py", line 137, in run
    env_runner = hydra.utils.instantiate(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 325, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET), full_key)
  File "/home/shi/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 139, in _resolve_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error locating target 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner', see chained exception above.
full_key: task.env_runner

Shi-jiayu, 2024-07-29:
When I train and evaluate a policy with behavior cloning or evaluate a saved policy, I will encounter this problem. May I ask what the problem is？Thanks！

YanjieZe, 2024-07-31:
ModuleNotFoundError: No module named 'natsort'

maybe try `pip install natsort`

closed 2024-08-01 by YanjieZe

abcdsaltfish, 2024-09-25:
I solve this problem by instantiate the env_runner in the train.py myself instead of hydra.

Replace ```dataset = hydra.utils.instantiate(cfg.task.dataset)```  with below
```python
from diffusion_policy_3d.dataset.adroit_dataset import AdroitDataset
dataset = AdroitDataset("/root/3D-Diffusion-Policy/3D-Diffusion-Policy/data/adroit_hammer_expert.zarr")
```

And then I installed natsort. However a new problem occured: the shape of matrixs are wrong. I don't how to continue.
```
Error executing job with overrides: ['task=adroit_hammer', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=adroit_hammer-dp3-0112', 'logging.mode=online', 'checkpoint.save_ckpt=True']
Traceback (most recent call last):
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 510, in <module>
    main()
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 507, in main
    workspace.run()
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 199, in run
    raw_loss, loss_dict = self.model.compute_loss(batch)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/policy/dp3.py", line 333, in compute_loss
    pred = self.model(sample=noisy_trajectory,
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/model/diffusion/conditional_unet1d.py", line 307, in forward
    x = resnet(x, global_feature)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/model/diffusion/conditional_unet1d.py", line 117, in forward
    embed = self.cond_encoder(cond)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/container.py", line 139, in forward
    input = module(input)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (128x256 and 384x1024)
Traceback (most recent call last):
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 510, in <module>
    main()
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
[... 5 lines cut ...]
    ret.return_value = task_function(task_cfg)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 507, in main
    workspace.run()
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/train.py", line 199, in run
    raw_loss, loss_dict = self.model.compute_loss(batch)
  File "/opt/data/private/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/policy/dp3.py", line 333, in compute_loss
    pred = self.model(sample=noisy_trajectory,
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
[... cut ...]
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/container.py", line 139, in forward
    input = module(input)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/mambaforge/envs/dpcopy/lib/python3.9/site-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (128x256 and 384x1024)
```

xiaolijz, 2024-10-15:
@abcdsaltfish I encountered the same problem, have you solved this problem?

xiaolijz, 2024-10-15:
@Shi-jiayu  I encountered the same problem, have you solved this problem?

slujiani, 2024-11-30:
I encounter the same problem, and I fix it with the command "export PYTHONPATH=$PYTHONPATH:/path-to DP3-repo/3D-Diffusion-Policy/third_party/dexart-release"

joanne40226, 2025-02-03:
@abcdsaltfish  @xiaolijz  @Shi-jiayu  @slujiani I encountered the same problem, have you solved this problem?

slujiani, 2025-03-11:
> I encounter the same problem, and I fix it with the command "export PYTHONPATH=$PYTHONPATH:/path-to DP3-repo/3D-Diffusion-Policy/third_party/dexart-release"

@joanne40226 I sovled the problem with the command 
`export PYTHONPATH=$PYTHONPATH:/path-to DP3-repo/3D-Diffusion-Policy/third_party/dexart-release`
You can change the path-to DP3-repo with your path, and try it. ^_^