YanjieZe/3D-Diffusion-Policy #60: Training dynamics

a-nooj, 2024-08-05:
Hi, thank you for open-sourcing your code.
I'm working on a similar project, integrating point cloud encoding with diffusion, and I'm curious about what your training curves looked like. Did you notice a steep decline in loss initially, followed by it flattening out?
I'm observing that the gradient means and standard deviations for my PointNet Encoder are very small, which makes me suspect it's not having much of a conditioning effect on the diffusion process. Did you notice anything similar?

a-nooj, 2024-08-07:
This is very helpful, I am currently investigating the loss curves for each diffusion step (instead of the aggregate loss) - will keep you updated with my progress!

Sorry if I missed this in the paper, but how long did it take to train for that last 1%?

YanjieZe, 2024-08-08:
Hi, I think generally we train 2000~3000 epochs. Observing from the loss, I could tell that in only several epochs (~10), the loss has been very small. but we usually still train for long to make our policy accurate

closed 2024-08-10 by YanjieZe