YanjieZe/3D-Diffusion-Policy #66: Is the point-cloud transformed to the robot arm base-link frame in the released real-world data ?

oym1994, 2024-09-03:
Hi , thanks for your great job.

I wander if the point-cloud in your released real-world data has been transformed to the robot arm base-link frame  ? 

Thanks for your kind response!

YanjieZe, 2024-09-03:
Hi, thank you for your interest.
I think in real-world exps, the point cloud is transformed from the camera frame into the world frame. So it does not matter whether the point cloud is in the robot arm frame.

closed 2024-09-03 by YanjieZe

oym1994, 2024-09-03:
> Hi, thank you for your interest. I think in real-world exps, the point cloud is transformed from the camera frame into the world frame. So it does not matter whether the point cloud is in the robot arm frame.

Thanks for your response!

I want to execute some experiments to prove whether the data you provided can by used to train for our UR5 robot arm.  One of the insight is that the point-cloud should be transformed to the robot arm base-link for I cant build the same robot config as yours. So is the point-cloud in your released real-world data transformed to the robot arm base-link frame ? If not, can you also provide the extrinsic between the camera and the robot arm?

Thanks again for your kind response!

YanjieZe, 2024-09-03:
I think it is not transformed to the robot arm base frame. Therefore I also do not have such transformation between the camera and the robot arm:) Sorry for the inconvenience. I think it could be not that easy for you to directly use our data on your UR5 robot. I could suggest collecting some new data, which would be much easier!