YanjieZe/3D-Diffusion-Policy #76: Question about using the data collection in realworld

Ju6276, 2024-10-16:
Thanks for your work! You mention in your repo that using the D435 camera will give a very poor point cloud. Is it a fusion of multiple D435's or just a single one?

YanjieZe, 2024-10-18:
We use the single camera. I think with multiple cameras and maybe some 3D reconstruction algorithms, D435 would be also helpful.

closed 2024-10-18 by YanjieZe

Ju6276, 2024-10-18:
Thank you for your reply. Your algorithms make a lot of sense, and I would like to reproduce the task pour and then train a task myself. So I prepared L515 camera, Franka robotic arm, Allegro hand. I am very much looking forward to the next step.
Now I have more questions about the setup. I noticed that you guys wrapped blue tape (or something?)  around the fingers, white tape where it connects to the Franka, and a blue mat. I would like to ask about the purpose of this tape here. (In my opinion, Is this for the camera to better recognise finger joints?)
Also is the model at the base of the Allgreo hand 3d printed in PLA material? I'm a bit worried if it's solid enough.
[image]

Thanks a lot for your time.