YanjieZe/3D-Diffusion-Policy #78: Questions about the experiment setup

Ju6276, 2024-10-18:
Your algorithms make a lot of sense, and I would like to reproduce the task pour and then train a task myself. So I prepared L515 camera, Franka robotic arm, Allegro hand. I am very much looking forward to the next step.
Now I have more questions about the setup. I noticed that you guys wrapped blue tape (or something?)  around the fingers, white tape where it connects to the Franka, and a blue mat. I would like to ask about the purpose of this tape here. (In my opinion, Is this for the camera to better recognise finger joints?)
Also is the model at the base of the Allgreo hand 3d printed in PLA material? I'm a bit worried if it's solid enough.
[image]

Thanks a lot for your time.

YanjieZe, 2024-10-20:
Hi, thank you for your interest.
The tape here is to simply to make our hands look nice:) The mat here is to fix the roll in case it slips. The mount is printed with PLA I think.

closed 2024-10-20 by YanjieZe

Ju6276, 2024-10-20:
Thanks for ur reply.