YanjieZe/3D-Diffusion-Policy #80: Error executing job with overrides

xiaolijz, 2024-10-24:
Hello, I encountered an error, `Error executing job with overrides: ['task=dexart_laptop', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=dexart_laptop-dp3-0322', 'logging.mode=online', 'checkpoint.save_ckpt=True']
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/jimmy/anaconda3/envs/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 503, in main
    workspace.run()
  File "train.py", line 142, in run
    assert isinstance(env_runner, BaseRunner)
NameError: name 'BaseRunner' is not defined`, How can I solve this problem?

YanjieZe, 2024-10-26:
Hi, this error is the first time to be reported. Have you solved?

xiaolijz, 2024-10-26:
> Hi, this error is the first time to be reported. Have you solved?

I solved this problem by running the Metaworld project first, and I found that the error disappeared. By the way, I’d like to know how to view the video of the results， I can’t find any videos in WandB.

YanjieZe, 2024-10-27:
Here we save the video. You can check the video file here (either save it to your local path, or check it on wandb)

https://github.com/YanjieZe/3D-Diffusion-Policy/blob/9fdfd0adcc7a9506118ad69b96ff7f4a425c8a40/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/metaworld_runner.py#L128

closed 2024-10-27 by YanjieZe