YanjieZe/3D-Diffusion-Policy #90: 三个问题，希望作者可以为我解答一下，衷心感谢

Corsimy, 2024-11-27:
1.训练和推理好像都只用到了点云，那为什么训练数据还需要彩色图和深度图呢？我在仿真环境生成训练数据时也可以直接sample世界坐标系下的点云，不需要图片的。希望您可以为我解答一下 谢谢。
2.假设n_obs_steps不为1，那在机器人启动时如何得到足够的观测呢？这时只有一个观测。我可以简单的把它复制为n_obs_steps的长度，然后启动推理吗？这个问题同样适用于2d 的 diffusion policy。希望您可以为我解答一下 谢谢。
3.我用dp3训练了‘adroit_hammer'，使用的是你们提供的数据，训练了3000个epoch，loss收敛的很小，eval了一下发现是如下的结果：
test_mean_score: 1.0                                                                                                                                                    
---------------- Eval Results --------------
mean_n_goal_achieved: 363.0000
mean_success_rates: 1.0000
test_mean_score: 1.0000
SR_test_L3: 1.0000
SR_test_L5: 1.0000
这些1.0是百分百成功了吗？以及363是在一共测试了多少次下的成功次数呢？这个结果看起来不错，但是当我打印模型推理的action时，action与gt_action差别很大：
Ground truth action: tensor([[[ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033]],

        [[ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 0.9998, -0.1457,  0.9998,  ...,  0.9997, -0.5486,  0.7629]],

        [[ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 1.0000, -0.1354,  0.9999,  ...,  0.9999, -0.2354,  0.3033],
         [ 0.9998, -0.1457,  0.9998,  ...,  0.9997, -0.5486,  0.7629],
         [ 0.3282, -0.5723,  0.9986,  ...,  0.9935, -0.9959,  0.6237]],

        ...,

        [[ 0.3599,  0.7440,  0.0309,  ..., -0.9914, -1.0000,  0.8647],
         [ 0.3810,  0.7460,  0.0495,  ..., -0.9918, -1.0000,  0.8356],
         [ 0.3725,  0.7567,  0.0470,  ..., -0.9927, -1.0000,  0.8227],
         [ 0.3629,  0.7644,  0.0363,  ..., -0.9931, -1.0000,  0.8130]],

        [[ 0.3810,  0.7460,  0.0495,  ..., -0.9918, -1.0000,  0.8356],
         [ 0.3725,  0.7567,  0.0470,  ..., -0.9927, -1.0000,  0.8227],
         [ 0.3629,  0.7644,  0.0363,  ..., -0.9931, -1.0000,  0.8130],
         [ 0.3738,  0.7618,  0.0322,  ..., -0.9932, -1.0000,  0.7944]],

        [[ 0.3725,  0.7567,  0.0470,  ..., -0.9927, -1.0000,  0.8227],
         [ 0.3629,  0.7644,  0.0363,  ..., -0.9931, -1.0000,  0.8130],
         [ 0.3738,  0.7618,  0.0322,  ..., -0.9932, -1.0000,  0.7944],
         [ 0.3636,  0.7662,  0.0174,  ..., -0.9928, -1.0000,  0.7845]]],
       device='cuda:0')
Predicted action: tensor([[[ 0.4319, -0.1392, -0.2543,  ...,  0.2854, -0.5419,  0.4181],
         [-0.1957,  0.3477,  0.7248,  ..., -0.0069, -0.6120, -0.3245],
         [ 0.0269, -0.6053, -0.4150,  ...,  0.2692, -0.4208,  0.2336],
         [-0.0867, -0.1685, -0.0622,  ..., -0.3003, -0.3626, -0.3263]],

        [[ 0.4256, -0.1434, -0.2394,  ...,  0.2715, -0.5362,  0.4173],
         [-0.1867,  0.3438,  0.7273,  ..., -0.0023, -0.6079, -0.3137],
         [ 0.0208, -0.6163, -0.3978,  ...,  0.2614, -0.4281,  0.2375],
         [-0.0855, -0.1694, -0.0758,  ..., -0.2893, -0.3639, -0.3267]],

        [[ 0.4332, -0.1304, -0.2028,  ...,  0.2534, -0.5347,  0.4120],
         [-0.1929,  0.3071,  0.6988,  ..., -0.0136, -0.6053, -0.2997],
         [ 0.0156, -0.6248, -0.3924,  ...,  0.2687, -0.4435,  0.2270],
         [-0.0783, -0.1483, -0.1034,  ..., -0.2777, -0.3533, -0.3122]],

        ...,

        [[ 0.2590, -0.1732, -0.2063,  ...,  0.4458, -0.4585,  0.4304],
         [ 0.1095,  0.4465,  1.0000,  ...,  0.1178, -0.4964, -0.0870],
         [-0.1032, -0.6505, -0.5161,  ...,  0.2355, -0.3275,  0.2013],
         [-0.1716, -0.1396,  0.1102,  ..., -0.2390, -0.3878, -0.4531]],

        [[ 0.2618, -0.1618, -0.1935,  ...,  0.4444, -0.4576,  0.4310],
         [ 0.1052,  0.4363,  1.0000,  ...,  0.1129, -0.4963, -0.0866],
         [-0.1117, -0.6547, -0.5153,  ...,  0.2169, -0.3254,  0.2070],
         [-0.1633, -0.1406,  0.1084,  ..., -0.2378, -0.3869, -0.4597]],

        [[ 0.2708, -0.1560, -0.1934,  ...,  0.4549, -0.4550,  0.4350],
         [ 0.1172,  0.4389,  1.0000,  ...,  0.1180, -0.4952, -0.0894],
         [-0.1062, -0.6516, -0.5121,  ...,  0.2023, -0.3268,  0.2124],
         [-0.1708, -0.1327,  0.1097,  ..., -0.2416, -0.3922, -0.4702]]],
       grad_fn=<ReshapeAliasBackward0>)
       他们明明用mse_loss优化过彼此之间的差异，loss也很小，为什么这里的差异这么大呢？
       
       最后附上我的测试代码：
       workspace = TrainDP3Workspace(cfg)

    # configure model
    workspace.model: DP3 = hydra.utils.instantiate(cfg.policy)

    workspace.ema_model: DP3 = None
    if cfg.training.use_ema:
        try:
            workspace.ema_model = copy.deepcopy(workspace.model)
        except: # minkowski engine could not be copied. recreate it
            workspace.ema_model = hydra.utils.instantiate(cfg.policy)

    device = torch.device(cfg.training.device)
    # configure dataset
    dataset: BaseDataset
    dataset = hydra.utils.instantiate(cfg.task.dataset)
    normalizer = dataset.get_normalizer()

    workspace.model.set_normalizer(normalizer)
    if cfg.training.use_ema:
        workspace.ema_model.set_normalizer(normalizer)
    
    if cfg.training.use_ema:
        mypolicy = workspace.ema_model
    else:
        mypolicy = workspace.model
    mypolicy.eval()

    assert isinstance(dataset, BaseDataset), print(f"dataset must be BaseDataset, got {type(dataset)}")
    val_dataloader = DataLoader(dataset, **cfg.val_dataloader)

                    
                    
    for i, batch in enumerate(val_dataloader):
        print(i)
        if i >= 10:
            break
        # Move batch to the device (GPU/CPU) as per configuration
        batch = dict_apply(batch, lambda x: x.to(device, non_blocking=True))
        
        # Extract observations and ground truth actions
        obs_dict = batch['obs']
        print("len of obs point cloud:  ",len(obs_dict['point_cloud']))
        print(obs_dict.keys())
        #sprint(obs_dict['point_cloud'])
        gt_action = batch['action']
       
        
        # Model predicts actions based on observations
        result = mypolicy.predict_action(obs_dict)['action_pred']
        actionn = mypolicy.predict_action(obs_dict)['action']
        # Print ground truth actions and model predictions
        print("Ground truth action:", gt_action)
        print("Predicted action:", result)
    
    衷心希望您可以为我解答一下，已经困扰我很久了，谢谢！！！

YanjieZe, 2024-11-27:
Hi, thank you for your interest.
1. You can directly use point clouds if you have. no need for depth/rgb.
2. stack the first frame twice
3. 1.0 means 100% success rate.

closed 2024-11-27 by YanjieZe

Corsimy, 2024-11-27:
@YanjieZe please . Thank you so much.

YanjieZe, 2024-11-27:
Could you maybe show the videos?

Corsimy, 2024-11-28:
here is a link of my  'adroit_hammer' demo displayed in wanb :
https://storage.googleapis.com/wandb-production.appspot.com/1799281482-Fudan%20University/dp3/32phs8be/media/videos/sim_video_eval_22406_276e02420021d737cef6.mp4?Expires=1732764709&GoogleAccessId=gorilla-files-url-signer-man%40wandb-production.iam.gserviceaccount.com&Signature=thfwIlOZjRZhxHsqGHhyl6B9iYjmMMPhXz%2BpakKn8nL00sXRoqFDwlP3cjBk3WbTgQ5wvPQp02fRjX4E1x2GkNNZRKMT0DfK8xM0QJoyJHJ0hp1j831t7%2Be%2FtMFZnObEW9yJbr6yha0VhFdJl%2Fsl1RXuOEQX8ft2Bz45YKrQaE3m84csMAYQrZFbcs1WBEFkZsiNzEX6MuSHjbruC26loNk0coF3rVr0LbtHjZg0TJ1RSaXPXFlKHP7b0IEV0mpITNCLbhlBxXMwBUvYHvXQmkQ4xvz9smGqZN4RHSrhDJmQUghK1Y9WrykBUeodHZ5QOvURH%2BPVZEdzcCiXwb6T6Q%3D%3D
@YanjieZe

YanjieZe, 2024-12-01:
Hi, this link can not be opened on my side. Could you give a new link?

Corsimy, 2024-12-02:
https://github.com/user-attachments/assets/187cceb4-80f2-46e1-97d5-7659348ef73f

@YanjieZe  thanks

Corsimy, 2024-12-10:
already solved. thanks @YanjieZe

qffof, 2025-01-08:
> 已经解决了。谢谢
你好，请问你解决了吗，我想请问一下为什么会出现这样的差距

Corsimy, 2025-01-08:
> > 已经解决了。谢谢
> > 你好，请问你解决了吗，我想请问一下为什么会出现这样的差距

因为我eval前没有正确load模型权重。哈哈

lanlankilkil, 2025-03-04:
> > > 已经解决了。谢谢
> > > 你好，请问你解决了吗，我想请问一下为什么会出现这样的差距
> 
> 因为我eval前没有正确load模型权重。哈哈

请问我在收集仿真演示数据时报错未注册怎么解决

Corsimy, 2025-03-04:
> > > > 已经解决了。谢谢
> > > > 你好，请问你解决了吗，我想请问一下为什么会出现这样的差距
> > 
> > 
> > 因为我eval前没有正确load模型权重。哈哈
> 
> 请问我在收集仿真演示数据时报错未注册怎么解决

我是在自己的仿真环境里实验的 你用的是什么环境？可以把报错复制给chatgpt问一下