YanjieZe/3D-Diffusion-Policy #93: questions about the pour real robot data and the application to franka arm

slujiani, 2024-11-30:
Hi, really excellent work! We have franka panda arm, and try to replay the pour task using the data given pour_40demo_1024.zarr.  There are the problems:

1、The RGB values of the point_cloud in some frames are negative, which causes an error when using `visualize_pointcloud` to visualize the point cloud.
```python
import zarr
import numpy as np
from pointcloud import visualize_pointcloud

zarr_data = zarr.open('data/pour_40demo_1024.zarr', mode='r')   

point_cloud_data = zarr_data['data/point_cloud'][1]

print(f"Point cloud data type: {type(point_cloud_data)}")
print(f"Point cloud data shape: {point_cloud_data.shape}")
print(f"Point cloud data: {point_cloud_data}")

point_cloud_data = np.array(point_cloud_data)  
visualize_pointcloud(point_cloud_data)
``` 
[image]
2、The last dimension of the action seems unusual, alternating between `-0.08500000089406967` and `0.08500000089406967`, except for when it is `0.0`. What does this value represent? From my understanding, `action[0:3]` corresponds to changes in the end-effector's position, while `action[3:6]` represents changes in the end-effector's Euler angles, am I right?

I use the following code to show the values of actions:
```python
import zarr
import numpy as np
from pointcloud import visualize_pointcloud
zarr_data = zarr.open('data/pour_40demo_1024.zarr', mode='r')   
action=zarr_data['data/action']
print(action.chunks)
print(f"action data type: {type(action)}")
print(f"action data shape: {action.shape}")
for i in action:
    print(f"action: {i[-1]}")
``` 
result：
[image]
after filtering out 0.0:
[image]

Besides, can I get some advice on robot controller? Should I use panda_py、 franka_ros or other implementations to conctrol franka arm?

slujiani, 2024-12-02:
I apologize for asking again, but could you please provide the data collection frequency, specifically how many frames per second? I'm concerned that I may not have set it correctly.

blakery-star, 2024-12-02:
Hi, thanks for your questions!
1. For visualization, I recommend using the provided visualization tool to visualize 'xyz', as our policy is trained with 'xyz' as input. There might be some incorrect RGB values in certain frames due to device issues. We will fix that.
2. The last dimension of the action represents the delta change in the gripper width. We use binary control for the gripper, so the value alternates between two fixed states.
3. You can use any common Franka controller; the ones you listed are acceptable. In our setup, we use Frankx as the Franka controller.
4. The demonstrations for the Franka arm were collected using a keyboard. As a result, we did not specify a collection frequency. You can choose an appropriate frequency to replay the data.

closed 2024-12-02 by blakery-star