YanjieZe/3D-Diffusion-Policy #97: Question about training！

Nikanbaturu, 2024-12-04:
尊敬的作者您好！十分感谢您无私的开源项目！但我在使用自己在真实环境中收集的数据进行训练时出现了以下报错：
Error executing job with overrides: ['task=real_put', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=real_put-dp3-0112', 'logging.mode=online', 'checkpoint.save_ckpt=True']
Traceback (most recent call last):
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/train.py", line 511, in <module>
    main()
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/train.py", line 508, in main
    workspace.run()
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/train.py", line 200, in run
    raw_loss, loss_dict = self.model.compute_loss(batch)
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/policy/dp3.py", line 314, in compute_loss
    pred = self.model(sample=noisy_trajectory,
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/model/diffusion/conditional_unet1d.py", line 308, in forward
    x = resnet(x, global_feature)
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/xwdrl/3D-Diffusion-Policy-master/3D-Diffusion-Policy/diffusion_policy_3d/model/diffusion/conditional_unet1d.py", line 118, in forward
    embed = self.cond_encoder(cond)
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/container.py", line 204, in forward
    input = module(input)
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/xwdrl/anaconda3/envs/dp3/lib/python3.10/site-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (128x256 and 384x1024)
这貌似是global_cond的温度与mlp输入不匹配造成的，我尝试修改yaml中的encoder_output_dim和diffusion_step_embed_dim，但发现无论如何调整，报错中的内容即mat1的第二个维度与 mat2的第一个维度始终相差一个obs_feature_dim。
请问要解决这个问题，需要进行什么样的修改？期待您的回答，在此感谢您的帮助！

YanjieZe, 2024-12-05:
What is dimension of your robo state? Have you modified the task config to make it suitable for your own setting?
[image]

Nikanbaturu, 2024-12-05:
感谢您的回答，我发现了问题所在。我使用hdf5文件来存储在实机上收集的数据，同时重写了dataset以在训练时加载数据集。貌似是这部分出现了错误，当我将数据集由hdf5转换为zarr后问题得以解决，可以正常训练了。

Nikanbaturu, 2024-12-05:
谢谢您的分享！

closed 2024-12-05 by Nikanbaturu

ruoxianglee, 2024-12-10:
> 感谢您的回答，我发现了问题所在。我使用hdf5文件来存储在实机上收集的数据，同时重写了dataset以在训练时加载数据集。貌似是这部分出现了错误，当我将数据集由hdf5转换为zarr后问题得以解决，可以正常训练了。

Hi, may I know how you convert hdf5 to zarr file? I would appreciate if you could share any link or code. Thanks.