YanjieZe/3D-Diffusion-Policy #98: Found 0 GPUs for rendering. Using device 0.

ggggfff1, 2024-12-05:
Hi,

I have encountered some problems with MetaWorld evaluation on a headless server. I suspect that MuJoCo is failing to find a GPU for rendering. However, I have confirmed that my GPUs work properly during training. Have you encountered this issue when deploying DP3 on a remote server? Looking forward to your reply!

Here is the error message:
> Found 0 GPUs for rendering. Using device 0.
> Device id outside of range of available devices.
> Traceback (most recent call last):
>   File "test.py", line 27, in <module>
>     img = sim.render(12, 12, camera_name='corner2', depth=False, device_id=-1,mode='offscreen',) 
>   File "mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
>   File "mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
>   File "mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
>   File "mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
>   File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
> RuntimeError: Failed to initialize OpenGL

Additionally, I referred to another [issue](https://github.com/YanjieZe/3D-Diffusion-Policy/issues/40), but it did not solve my problem.

Thank you for your help!

YanjieZe, 2024-12-05:
This seems to be a problem in metaworld/mujoco/opengl. like this?
https://github.com/openai/mujoco-py/issues/187

wut19, 2024-12-06:
@ggggfff1 Have you handled this problem? I've met the same problem. Any suggestions? 
Thank you.

ggggfff1, 2024-12-09:
> @ggggfff1 Have you handled this problem? I've met the same problem. Any suggestions? Thank you.

Hi, I found a solution, is to use
```
self.viewer = mujoco_py.MjRenderContextOffscreen(self.env.sim, None, device_id=self.device_id) 
self.viewer.render(self.image_size, self.image_size, self.env.sim.model.camera_name2id("corner2"))
img = self.viewer.read_pixels(self.image_size, self.image_size, depth=False)
```
instead of 
`img = self.env.sim.render(width=self.image_size, height=self.image_size, camera_name="corner2", device_id=self.device_id)`
in 3D-Diffusion-Policy/diffusion_policy_3d/env/metaworld/metaworld_wrapper.py line130

but it will occur new problems. The eval speed will become slower and the result will become bad. @wut19

ggggfff1, 2024-12-09:
> This seems to be a problem in metaworld/mujoco/opengl. like this? [openai/mujoco-py#187](https://github.com/openai/mujoco-py/issues/187)

I have tried this method, but it doesn't work. @YanjieZe

euminds, 2025-08-24:
> > This seems to be a problem in metaworld/mujoco/opengl. like this? [openai/mujoco-py#187](https://github.com/openai/mujoco-py/issues/187)
> 
> I have tried this method, but it doesn't work. [@YanjieZe](https://github.com/YanjieZe)

So what is the effective solution now?