YanjieZe/3D-Diffusion-Policy #99: "wrappers.pxi", line 1071, in mujoco_py.cymj.PyMjModel.camera_name2id ValueError: No "camera" with name top exists. Available "camera" names = ('fixed', 'vil_camera').

xzhuzhu, 2024-12-08:
Hi, thanks for your great works.
When I am running the code "bash scripts/gen_demonstration_adroit.sh hammer
", I have encountered the following problem:
File "wrappers.pxi", line 1071, in mujoco_py.cymj.PyMjModel.camera_name2id
ValueError: No "camera" with name top exists. Available "camera" names = ('fixed', 'vil_camera').

How can I solved it? The mujoco_py version I used is mujoco_py-2.1.2.14.

YanjieZe, 2024-12-08:
Hi, It says no such camera. Are you using the adroit environment we provide? or do you maybe do some modifications on the code?

xzhuzhu, 2024-12-10:
I created a new environment and followed the instructions outlined in the 'INSTALL.md' file to set it up. I did not make any modifications to the process or the environment itself. However, I encountered the issue during the setup. Could you provide some suggestions on how to resolve this problem?

YanjieZe, 2024-12-10:
Yes. it shows you do not have such camera, so you just need to change a camera that you are using.

closed 2024-12-10 by YanjieZe