YanjieZe/3D-Diffusion-Policy #103: Question about generate data?

hschu11, 2024-12-18:
Hello, thanks for your excellent work!
For the MetaWorld tasks, according to the expert model you provided, the `push_back` and `handle-press-side` tasks ran for 20 hours without successfully generating any data, while other tasks were able to construct data normally.
[image]

YanjieZe, 2024-12-18:
Hi, thank you for your interest.
You can see that the success times are larger than 0, which means it succeeds. You can change the required success times in the code to be a smaller number, so that you can generate data successfully.

closed 2024-12-18 by YanjieZe

hschu11, 2024-12-19:
> Hi, thank you for your interest. You can see that the success times are larger than 0, which means it succeeds. You can change the required success times in the code to be a smaller number, so that you can generate data successfully.

Thank you for your reply. By modifying the `success_time`, the `handle-press-side` data can be generated, but the `push-back` task still cannot be generated because its `success_time` is always 0.
[image]