YanjieZe/3D-Diffusion-Policy #106: Problem about action

Haiyuan-Liu, 2024-12-28:
Hello Yanjie,

Could you please explain the action you have used?
As you mentioned, you used relative end-effector position control for the robot arm and relative joint-angle position control for the dex hand. I still feel confused about the "relative end-effector position": Does that mean the *change* of position and orientation in global coordinates between the previous frame and the current frame? 

Looking forward to your response!
Thank you very much!

Haiyuan-Liu, 2024-12-28:
This is my understanding of "relative end-effector position control" mentioned in the ReadMe:

Assume the position of end-effector in the global coordinate at time step $t$ is $p_t$, 
the position of end-effector at next time step is $p_{t+1}$, 
then the action $a_t$ at time step $t$ is $p_ {t+1} - p_t$.

Is it correct?

YanjieZe, 2024-12-31:
```
Does that mean the change of position and orientation in global coordinates between the previous frame and the current frame?
```
Yes, I think so.

closed 2024-12-31 by YanjieZe