YanjieZe/3D-Diffusion-Policy #108: quetion about scaling up.

jinyqjinyq, 2025-01-04:
hello，I encountered a problem in my recent experiment and hope to get your answer. For the same task, after training with 100 pieces of data, the test has a relatively good success rate, but after training with 300 pieces of data, the success rate is greatly reduced. I wonder what causes this situation?

YanjieZe, 2025-01-06:
This is interesting. If the data distribution is right, the policy usually monotonically improves. So maybe check the data first.

closed 2025-01-06 by YanjieZe

jinyqjinyq, 2025-01-07:
> This is interesting. If the data distribution is right, the policy usually monotonically improves. So maybe check the data first.
yes, i have check the data, confirm that there are no issues with data distribution. Maybe we can discuss this issue further。