YanjieZe/3D-Diffusion-Policy #109: Assistance Needed: Segmentation Fault in gen_demonstration_adroit.sh Execution

beginner-123, 2025-01-05:
Hello Yanjie,
I am currently encountering an issue while running the following command:bash scripts/gen_demonstration_adroit.sh hammer  

The terminal outputs the following message:
Found 3 GPUs for rendering. Using device 0.  
scripts/gen_demonstration_adroit.sh: line 15: 28299 Segmentation fault  

However, my system only has a single NVIDIA RTX 4090 GPU.

After debugging, I traced the issue to the following line of code in third_party/VRL3/src/rrl_local/rrl_multicam.py (line 115):
img = self._env.env.sim.render(width=self.width, height=self.height, mode='offscreen', camera_name=cam) # TODO device id will think later  
When this line is executed, the program unexpectedly exits with a segmentation fault.

I would appreciate your assistance with the following:
    Understanding why the program detects multiple GPUs when only one is available on my system.
    Configuring the sim.render function to ensure it works correctly on my single GPU setup.

closed 2025-01-06 by YanjieZe

hiskiv, 2025-01-11:
Same segmentation fault issue here, despite I am running on a 8 gpu server.

HilbertXu, 2025-01-12:
The same segmentation fault issue here, tried on two PCs, both of them are ubuntu 20.04 with CUDA-11.8.

HilbertXu, 2025-01-13:
Greetings! @hiskiv @YanjieZe @beginner-123 

finally I got this issue solved with a new and clean ubuntu 20.04 system with cuda 11.7. 
I failed on two ubuntu 20.04 machine with cuda 11.8, so I would suspect that this issues is stemed from cuda, cuz the mujoco-py will be compiled with cython and nvcc at the first time when it is imported.

Basically, I created a python3.8 env with mamba and installed pytorch 13.1+cu117 with pip. Then you might need to follow the instruction in [here](https://github.com/openai/mujoco-py/blob/master/Dockerfile) to install some system requirements for mujoco_py, then just follow the install instruction of DP3. This should work on a new and clean ubuntu 20.04 system. 

For anyone who don't want to reinstall the system, I also wrapped a docker image for this repo, you can find it [here](https://hub.docker.com/r/hilbertxuyc/dp3-cuda11.7-mamba). I have tested it for generating demons as well as training policy.

Cheers,
Yucheng