YanjieZe/3D-Diffusion-Policy #112: issue when training a policy with example instructions

joanne40226, 2025-02-03:
hi, thank you for the excellent work!
i followed the instructions and finished generating demonstrations by gen_demonstration_adroit.sh
`bash scripts/gen_demonstration_adroit.sh hammer`

when i tried to train the policy by
`bash scripts/train_policy.sh dp3 adroit_hammer 0112 0 0`

something went wrong.

it can show these successfully without error

```
gpu id (to use): 0
Train mode
/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'simple_dp3.yaml': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
[DP3Encoder] point cloud shape: [512, 3]
[DP3Encoder] state shape: [24]
[DP3Encoder] imagination point shape: None
[PointNetEncoderXYZ] use_layernorm: True
[PointNetEncoderXYZ] use_final_norm: layernorm
[DP3Encoder] output dim: 128
[SDP3] use_pc_color: False
[SDP3] pointnet_type: pointnet
[2025-02-03 15:47:39,454][diffusion_policy_3d.model.diffusion.simple_conditional_unet1d][INFO] - number of parameters: 7.394970e+06
----------------------------------
Class name: ConditionalUnet1D
  Number of parameters: 7.3950M
   mid_modules: 1.7725M (23.97%)
   diffusion_step_encoder: 0.1317M (1.78%)
   up_modules: 2.6089M (35.28%)
   down_modules: 2.7962M (37.81%)
   final_conv: 0.0857M (1.16%)
----------------------------------
----------------------------------
Class name: SimpleDP3
  Number of parameters: 7.4598M
   _dummy_variable: 0.0000M (0.00%)
   obs_encoder: 0.0648M (0.87%)
   model: 7.3950M (99.13%)
   mask_generator: 0.0000M (0.00%)
----------------------------------
Replay Buffer: state, shape (1000, 24), dtype float32, range -1.21~1.60
Replay Buffer: action, shape (1000, 26), dtype float32, range -1.00~1.00
Replay Buffer: point_cloud, shape (1000, 512, 6), dtype float32, range -0.25~255.00
Replay Buffer: img, shape (1000, 84, 84, 3), dtype uint8, range 0.00~255.00
--------------------------

```

However, the following shows that there are problems about **ImportError: Error loading 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner':
AttributeError("module 'sapien.core' has no attribute 'VulkanRenderer'")**
and also
**hydra.errors.InstantiationException: Error locating target 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner', see chained exception above.
full_key: task.env_runner**

the detailed output are as follow

```
root@b7104d146c75:/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy# bash scripts/train_policy.sh simple_dp3 adroit_hammer 0322 0 0
gpu id (to use): 0
Train mode
/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'simple_dp3.yaml': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
[DP3Encoder] point cloud shape: [512, 3]
[DP3Encoder] state shape: [24]
[DP3Encoder] imagination point shape: None
[PointNetEncoderXYZ] use_layernorm: True
[PointNetEncoderXYZ] use_final_norm: layernorm
[DP3Encoder] output dim: 128
[... 93 lines cut ...]
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 514, in main
    workspace.run()
  File "train.py", line 139, in run
    env_runner = hydra.utils.instantiate(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 222, in instantiate
    return instantiate_node(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 325, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET), full_key)
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 139, in _resolve_target
    raise InstantiationException(msg) from e
hydra.errors.InstantiationException: Error locating target 'diffusion_policy_3d.env_runner.adroit_runner.AdroitRunner', see chained exception above.
full_key: task.env_runner
```

thank you in advanced for the support.

YanjieZe, 2025-02-03:
```
AttributeError: module 'sapien.core' has no attribute 'VulkanRenderer'
```
Hi, this seems to be the issue of Sapien. Could you try with other Sapien's version?
Besides, are you running on a ubuntu machine with GPU?

joanne40226, 2025-02-04:
@YanjieZe 
Thank you so much for your reply!

Yes as I upgrade Sapien to the latest version, it worked now! Thank you.
However, after I log in the wandb, 
it showed the error as follow and I'm not sure is that related to GPU issue.

`Eval in Adroit hammer Pointcloud Env:   0%|                                                                                                                                         | 0/20 [00:00<?, ?it/s]Found 3 GPUs for rendering. Using device 0.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
Could not initialize EGL
Error executing job with overrides: ['task=adroit_hammer', 'training.debug=False', 'training.seed=0', 'training.device=cuda:0', 'exp_name=adroit_hammer-simple_dp3-0322', 'logging.mode=online', 'checkpoint.save_ckpt=True']
Traceback (most recent call last):
  File "train.py", line 518, in <module>
    main()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/main.py", line 90, in decorated_main
    _run_hydra(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
    _run_app(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 452, in _run_app
    run_and_report(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
    raise ex
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
    return func()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
    lambda: hydra.run(
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/dp3/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 515, in main
    workspace.run()
  File "train.py", line 263, in run
    runner_log = env_runner.run(policy)
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env_runner/adroit_runner.py", line 74, in run
    obs = env.reset()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/multistep_wrapper.py", line 127, in reset
    obs = super().reset()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/gym-0.21.0/gym/core.py", line 292, in reset
    return self.env.reset(**kwargs)
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/video_recording_wrapper.py", line 23, in reset
    obs = super().reset(**kwargs)
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/third_party/gym-0.21.0/gym/core.py", line 292, in reset
    return self.env.reset(**kwargs)
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/mjpc_diffusion_wrapper.py", line 184, in reset
    obs_dict = self.env.reset()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/adroit/adroit.py", line 340, in reset
    obs_pixels, obs_sensor = self._env.reset()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/adroit/rrl_local/rrl_multicam.py", line 162, in reset
    pixels, sensor_info = self.get_obs()
  File "/home/os-joanne.chung/src/robotLearning/src/3D-Diffusion-Policy/3D-Diffusion-Policy/diffusion_policy_3d/env/adroit/rrl_local/rrl_multicam.py", line 114, in get_obs
    img = self._env.env.sim.render(width=self.width, height=self.height, mode='offscreen', camera_name=cam, device_id=0) # TODO device id will think later
  File "mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
  File "mjrendercontext.pyx", line 46, in mujoco_py.cymj.MjRenderContext.__init__
  File "mjrendercontext.pyx", line 114, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
wandb: 
wandb: 🚀 View run 0 at: https://wandb.ai/imjoannne-national-taiwan-university/dp3/runs/6ke6t2z2
wandb: Find logs at: data/outputs/adroit_hammer-simple_dp3-0322_seed0/wandb/run-20250204_083722-6ke6t2z2/logs
`

Again, I appreciate your help in advanced. Thank you so much.

joanne40226, 2025-02-04:
@YanjieZe  Yes as I upgrade Sapien to the latest version, it worked now! Thank you so much!!

joanne40226, 2025-02-04:
@YanjieZe Hi, I also wonder if i can customize the dataset format myself with the DP3 policy.
Instead of the 3D point cloud, i also have the force data for model input.
So I wonder if it is possible to slightly change the input and output.

thank you in advanced for the support again!

YanjieZe, 2025-02-04:
Glad that the problem is solved.
Sure, you can modify the training data as you want, and you also need to modify the network acrh to take the changed data. In this case you need to change the pointnet's input from Nx6 to Nx7. See this file: https://github.com/YanjieZe/3D-Diffusion-Policy/blob/master/3D-Diffusion-Policy/diffusion_policy_3d/model/vision/pointnet_extractor.py

closed 2025-02-04 by YanjieZe

joanne40226, 2025-02-04:
@YanjieZe i see, i'll try then. thank you so much for answering!

abhishek47kashyap, 2025-02-10:
I too am running on Ubuntu (22.04) with GPU and my `sapien` version is `2.2.1`, and am running into the problem
```
AttributeError: module 'diffusion_policy_3d.env_runner' has no attribute 'adroit_runner'
```

Which version of `sapien` actually works? I've already checked `2.2.2`, `2.1.0`, `1.1.1`, `1.1.0` ([release history](https://pypi.org/project/sapien/#history))

Ju6276, 2025-04-07:
> [@YanjieZe](https://github.com/YanjieZe) i see, i'll try then. thank you so much for answering!

HI. I'm also trying to fuse force and vision. Have you been successful in achieving this and how well did it work?

Ju6276, 2025-04-07:
> [@YanjieZe](https://github.com/YanjieZe) Hi, I also wonder if i can customize the dataset format myself with the DP3 policy. Beside the 3D point cloud, i also have the force data for model input. So I wonder if it is possible to slightly change the input with adding one more data - force data so that the point cloud will be (T, Np, 6+1) where Np is the number of point clouds, 6 denotes [x, y, z, r, g, b, force]
> 
> thank you in advanced for the support again!

HI. I'm also trying to fuse force and vision. Have you been successful in achieving this and how well did it work?