YanjieZe/3D-Diffusion-Policy #113: About the example trained policies and deployment code.

joanne40226, 2025-02-05:
Hi, thank you so much for the great work.

When i was looking at the README file, it said that for deployment code we should refer iDP3 for a reference.
However, when i checked [iDP3](https://github.com/YanjieZe/Improved-3D-Diffusion-Policy), i found that the deployment code is for Humanoid Teleoperation.

I wonder if there are any example trained policies for robot arm such as Franka.
Also, i wonder if there are any deployment code for robot arm?

Thank you for your great work and thank you for your support in advanced!

joanne40226, 2025-02-05:
@YanjieZe Sorry to bother. But also, when i check the training__data_example, i found that the shape of action is action shape: (541, 25), which the action dim = 25.
I'm not sure why since i thought it should only be 6d of the end effector of the 2 fingered gripper positions.
Further more, when i tired raw_grasp data, i think it did not collect the color data so it cannot be convert to the desired format.

YanjieZe, 2025-02-28:
Hi, thank you for your interest.
Currently this part of code is not cleaned yet; but we have released related instructions in readme for this part, and I believe the deployment code in iDP3 would be helpful also. Thank you for your understanding.

closed 2025-02-28 by YanjieZe