YanjieZe/3D-Diffusion-Policy #114: How to create an aligned point cloud from multiple cameras in Metaworld?

Trevor-emt, 2025-02-23:
Dear author,
    I slightly change your source code to generate the point cloud from serval cameras like setting cam_names=['corner2', 'corner', 'corner3'] for Metaworld. But I found the generated point cloud is not aligned, i was wondering if you have met the same problem and finally utilize only one camera to form the point cloud. Below is the generated point cloud in one of the environment with camera names being ['corner2', 'corner', 'corner3']:

[image]

Best regards, 
Trevor-emt

YanjieZe, 2025-02-28:
The answer above looks great.
Yes I think the transformation in our original code is somehow problematic. But since we only use single view, it is fine.
To use multi views, need some fix like @b-karthiknair mentioned.
Thanks!

closed 2025-02-28 by YanjieZe

joonsu0109gh, 2025-03-31:
Thanks for the great comments!
I'm working on generating wrist-view point clouds, and your feedback helped me a lot.
By the way, could I ask the authors about the purpose of MetaWorld.pc_transform that [b-karthiknair](https://github.com/b-karthiknair) mentioned?

jangyicheng, 2025-06-11:
The above code is only suitable for static cameras. Below I provide code for dynamic camera point cloud alignment that works in any MuJoCo environment:

                cam_id = self.sim.model.camera_name2id(self.cam_names[cam_i])
                
                cam_body_id = self.sim.model.cam_bodyid[cam_id]
                cam_pos = self.sim.data.cam_xpos[cam_id]
                c2b_r = rotMatList2NPRotMat(self.sim.data.cam_xmat[cam_id])#self.sim.data.cam_xmat[cam_id]
                b2w_r = quat2Mat([0, 1, 0, 0])
                c2w_r = np.matmul(c2b_r, b2w_r)
                c2w = posRotMat2Mat(cam_pos, c2w_r)