YanjieZe/3D-Diffusion-Policy #118: Question about fps and muti-cameras

LogSSim, 2025-03-05:
Thanks for your contributions to this great work. I want to know how to get the original points cloud and multi-camera data.

YanjieZe, 2025-03-06:
Hi, thank you for your interest. 
See here how we get poin cloud and process it: https://github.com/YanjieZe/3D-Diffusion-Policy/blob/c72a1ace81d1217f1e6450ca7a98fcd3b668c009/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/mjpc_wrapper.py#L144C9-L144C24

closed 2025-03-06 by YanjieZe

LogSSim, 2025-03-11:
> Hi, thank you for your interest. See here how we get poin cloud and process it: https://github.com/YanjieZe/3D-Diffusion-Policy/blob/c72a1ace81d1217f1e6450ca7a98fcd3b668c009/3D-Diffusion-Policy/diffusion_policy_3d/gym_util/mjpc_wrapper.py#L144C9-L144C24

Ok, Thanks.