YanjieZe/3D-Diffusion-Policy #122: some questions about training results

parkergu, 2025-03-22:
Hi,

Thank you very much for your excellent work. I have a few questions regarding training dp3 using the dexart dataset:

    Is it the case that the outcomes of each training are different?

    Does the checkpoint with the highest success rate during training necessarily perform best during testing?

    Is it possible that training on different GPUs may yield inconsistent results?

I look forward to your kind response.

YanjieZe, 2025-03-22:
1. I think the training results (e.g., checkpoints) should be very stable, but evaluation   in simulation has randomness, which makes your "training results" different. This is common in simulation.
2. Not necessarily
3. Possibly. training on A100 might be different from training on A40, or etc. But it should not have much difference. These differences are pretty negligible.

closed 2025-03-22 by YanjieZe

parkergu, 2025-03-22:
Thank you. It is helpful.