YanjieZe/3D-Diffusion-Policy #126: [Question] train_policy.sh

Frozenkiddo, 2025-04-22:
Hello, I am using this repo by following the README.md, and I wanna know that shat do the following numbers refer to:
```
bash scripts/train_policy.sh dp3 adroit_hammer 0112 0 0
```
If I train on different envs or tasks, should I use the same num? @YanjieZe

SuperFanBin, 2025-04-24:
@Frozenkiddo You don't have to use the same num. The first number is an additional label for recording this training session, the second number is used to set the random seed, and the last number indicates the GPU serial number to be used.

Frozenkiddo, 2025-04-24:
Thank you for your reply! Does the addstional label '0112' have certain meaning? @SuperFanBin
> [@Frozenkiddo](https://github.com/Frozenkiddo) You don't have to use the same num. The first number is an additional label for recording this training session, the second number is used to set the random seed, and the last number indicates the GPU serial number to be used.

SuperFanBin, 2025-04-24:
No. It is just used to help you distinguish between different experiments. I guess 0112 means January 12🤔

Frozenkiddo, 2025-04-24:
OK, that helps a lot. Thanks again for your reply! @SuperFanBin