YanjieZe/3D-Diffusion-Policy #141: push-back dataset

LogSSim, 2025-05-15:
Thanks for your great work!! I want to know how I can generate push-back data. 

[image]

YanjieZe, 2025-05-15:
Sometimes it happens. Could you visualize the video to see what is happening?

LogSSim, 2025-05-17:
> Sometimes it happens. Could you visualize the video to see what is happening?

How can I visualize and generate a demo of the data?

YanjieZe, 2025-05-22:
You can check the script you use to generate data and use the saved image sequence to save it as a video.
for example, you can see here: https://github.com/YanjieZe/3D-Diffusion-Policy/blob/4157303813c4968fd15a241ac07c6d088b574d3a/third_party/dexart-release/examples/gen_demonstration_expert.py#L200
There is a stack of images. Saving it as the video will help you see what demo look like.

closed 2025-05-22 by YanjieZe