YanjieZe/3D-Diffusion-Policy #148: Realdex_Drill State-Action Pair Numerical Problems

Ethonwinters, 2025-06-03:
Hi ,Thanks for your great work!!
I have some confusion in State-Action Pair Numerical
at time t state[t+1]=state[t]+action[t] is right?
but the following array is not like this 
eg:  state[0,2] +action[0,2] =state[1,2]    0.291528 + (-0.045) =  0.246528
       but state[1,2] +action[1,2] not equal state[2,2]    0.246528 + (-0.045)  not equal to 0.246528

state=array([[ 0.549424  , -0.226859  ,  0.291528  , -0.415828  ,  0.966995  ,1.40781   , -0.47  ,  0.21294071,  0.11387723,  0.26207733,-0.31450114,  0.23472819, -0.01148471,  0.14734478,  0.47      ,
               0.21670218,  0.06802595,  0.214481  ,  1.3514967 ,  0.10917508,0.85332996,  0.20835514],
               
             [ 0.549424  , -0.226859  ,  0.246528  , -0.415828  ,  0.966995  ,1.40781   , -0.47      ,  0.21294071,  0.11387723,  0.26207733, -0.31450114,  0.23472819, -0.01148471,  0.14734478,  0.47      ,
               0.21670218,  0.06802595,  0.214481  ,  1.3514967 ,  0.10917508,0.85332996,  0.20835514],
               
             [ 0.549424  , -0.226859  ,  0.246528  , -0.415828  ,  0.966995  ,1.40781   , -0.47      ,  0.21294071,  0.11387723,  0.26207733,-0.31450114,  0.23472819, -0.01148471,  0.14734478,  0.47      ,
               0.21670218,  0.06802595,  0.214481  ,  1.3514967 ,  0.10917508,0.85332996,  0.20835514],
               
             [ 0.504424  , -0.226859  ,  0.246528  , -0.415828  ,  0.966995  ,1.40781   , -0.47      ,  0.21294071,  0.11387723,  0.26207733,-0.31450114,  0.23472819, -0.01148471,  0.14734478,  0.47      ,
               0.21670218,  0.06802595,  0.214481  ,  1.3514967 ,  0.10917508,0.85332996,  0.20835514],
               
             [ 0.459424  , -0.226859  ,  0.246528  , -0.415828  ,  0.966995  ,1.40781   , -0.47      ,  0.21294071,  0.11387723,  0.26207733,-0.31450114,  0.23472819, -0.01148471,  0.14734478,  0.47      ,
               0.21670218,  0.06802595,  0.214481  ,  1.3514967 ,  0.10917508,0.85332996,  0.20835514],], dtype=float32)

action=array([[ 0.   ,  0.   , -0.045,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ],
              [ 0.   ,  0.   , -0.045,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ],
              [ 0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ],
              [-0.045,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ],
              [-0.045,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ,  0.   ],], dtype=float32)

YanjieZe, 2025-06-24:
Hi, thank you for your interest!
I think during data collection we do not strictly follow state[t+1]=state[t]+action[t]. Let me know if you have any problem on your own robot.

closed 2025-06-24 by YanjieZe

Ethonwinters, 2025-06-25:
> Hi, thank you for your interest! I think during data collection we do not strictly follow state[t+1]=state[t]+action[t]. Let me know if you have any problem on your own robot.

Thank you for your replay,I want to know the relation between state[t+1],state[t],action[t] at time step t of dataset.