YanjieZe/3D-Diffusion-Policy #150: Help with DP3 performance on Robomimic 'can' and 'square' tasks

jangyicheng, 2025-06-11:
I am writing to seek some guidance regarding DP3's performance on several Robomimic tasks.

I have been experimenting with DP3 and observed a significant performance discrepancy. While it achieves a 100% success rate on the lift task, its performance on the can and square tasks is near 0%, which is substantially lower than the results from the original Diffusion Policy. Furthermore, I've noticed that the convergence of the success rate during training is considerably slower compared to the baseline.

My primary suspicion is that this may be related to the point cloud processing, specifically the cropping range. My current setup utilizes point clouds from two camera perspectives: agentview and robot0_eye_in_hand.

I suspect this might be related to the point cloud cropping range. What would be the most effective cropping parameters? Currently, I'm using point clouds from two perspectives: 'agentview' and 'robot0_eye_in_hand' . Additionally, how can we determine whether the point cloud information is sufficient for the robot to successfully complete the tasks? 

Any insights or guidance you could provide would be greatly appreciated.

Thank you for your time and consideration！

Below are my point cloud

can:
[image]

square
[image]

[image]

[image]

YanjieZe, 2025-06-24:
Hi, thank you for your interest!
This is an interesting phenomenon. I do not know clearly why it has only 0% success rate. Could you show some evaluation videos? like failure cases of DP3

jangyicheng, 2025-06-25:
Appreciate your feedback!

Video analysis revealed two key failure modes:​

Can and Square task failures mainly occur from empty grasps (wrong gripping positions)
​​Even when grasping succeeds, the Square task may still place the object in the wrong position.

It appears that when there are issues with my point cloud processing, the lift operation is unlikely to succeed. What's particularly puzzling is that for type-identical tasks(pick), the lift task achieves a 100% success rate without position error,while the can task and square task consistently fails to grasp the object.

Thank you for your help!

Below are the video demonstrations for the two tasks respectively:

task lift, success_rate = 1.0:

https://github.com/user-attachments/assets/aa6a283e-7d4c-46c0-856a-7bbbb9fe9af7

task can, success_rate = 0:

https://github.com/user-attachments/assets/272e5103-fa61-4c09-9f1f-bed629b7c14e

https://github.com/user-attachments/assets/8872e35e-b715-4dc2-ae0c-8dacc7a29712

task square, success_rate = 0:

https://github.com/user-attachments/assets/3b03ff9d-c832-41cf-852d-309822230314

https://github.com/user-attachments/assets/0cac879e-fea7-4182-9e29-aeb1650477a7

YanjieZe, 2025-06-25:
It feels like the object's point clouds are not clear to the policy; i guess maybe because of the points from the table? Or maybe add color information to make the object points more distinguished?

jangyicheng, 2025-06-26:
After using RGB point clouds, the success rate still hasn't improved.  

What puzzles me is:  
1. The model can achieve fitting on the training set, but in the validation environment with consistent point cloud patterns, it always grasps at obviously incorrect positions.  

2. The model achieves a 100% grasping success rate on relatively simple lifting tasks, but on the "can" and "square" tasks, it almost always grasps at completely wrong positions.  

3. In the point cloud of the square task shown in the figure below, the object's position in the z-direction is obviously incorrect, yet there are no such errors in the point clouds of other tasks.

https://github.com/user-attachments/assets/621950f1-e8f6-45ef-ac3a-9d09fe9eb379

https://github.com/user-attachments/assets/2f3d668e-0939-404f-88c3-c926d1463c09

The left side shows the point cloud from the training data, while the right side displays the point cloud during validation.

[image]

[image]

YanjieZe, 2025-06-29:
Closing this issue. For new questions welcome to open new issue.

xueking1, 2025-09-21:
Hello, may I ask which simulator you used to generate these point clouds? Thank you

jangyicheng, 2025-09-30:
> Hello, may I ask which simulator you used to generate these point clouds? I hope to receive your reply. Thank you!

mujoco